platformio/platformio-core #2367: Import project from Keil µVision IDE

ivankravets, 2019-04-22:
See https://community.platformio.org/t/port-keil-project-to-platformio/7409

Implement `platformio project import` command.

For example,
```
platformio project import [KEIL_PROJECT_DIR] -d,--destination-dir [WRITABLE_DIR] -t,--type [keil|...]
```

deladriere, 2019-10-14:
Any update on this feature ? I am trying to move a project from Keil to PIO but I am getting tons of errors

jwiem, 2020-03-16:
Has anyone found a workaround for this?

dkgdsc, 2021-04-02:
Hello! I am a student from China, I am responding to the problem through the translator, because my English is not good, please forgive me! Why do I encounter the following error when importing keil to build the STM32 standard library project: Traceback (most recent call last): File "/usr/local/bin/platformio", line 5, in from platformio.__main__ import main ModuleNotFoundError: No module named'platformio'

TianchouYin, 2022-07-18:
> 因为这个导入工程是导入之前用platformio创建的工程

zhiyuanzhai, 2022-10-22:
Any more progress on this new feature?

Andy-Cheng1017, 2024-12-10:
EIDE have this feature, Why platformio still not achive?

WalterTrovo, 2024-12-10:
Same problem here... I have tons of project i would like to import in PIO... Any step forward?