platformio/platformio-core #2369: Home: Could not load PIO Account information

Penguone, 2019-04-22:
PIO Core Call Error: "Details: http://docs.platformio.org/page/userguide/account/index.html\n\nYou are not logged in. Please log in to PIO Account using PlatformIO IDE > PIO Home > Account or `pio account login` command and try again. \n\nIf you don't have PIO Account yet, please create it using PlatformIO IDE > PIO Home > Account or `pio account register` command."

ivankravets, 2019-04-22:
Please update PIO Core packages via:

a) Open PlatformIO IDE Terminal and type `pio update`
b) PlatformIO IDE > PlatformIO Activity Bar (left sidebar) > Updates > Update All

Restart VSCode.

ivankravets, 2019-04-23:
Fixed in the latest PIO Core 3.6.7. Please open PlatformIO IDE Terminal and type `pio upgrade`. Restart VSCode.

closed 2019-04-23 by ivankravets