platformio/platformio-core #2408: Home: Could not initialize project

IshanBhavsar, 2019-04-24:
Server error: {"message":"[Error 2] The system cannot find the file specified: u'C:\\\\Users\\\\Ishan\\\\Documents\\\\PlatformIO'","args":[2,"The system cannot find the file specified"],"type":"WindowsError"}

IshanBhavsar, 2019-04-24:
trying to create my first project

IshanBhavsar, 2019-04-25:
PIO Core Call Error: "The next files/directories have been created in C:\\Users\\Ishan\\Documents\\PlatformIO\\Projects\\test\r\ninclude - Put project header files here\r\nlib - Put here project specific (private) libraries\r\nsrc - Put project source files here\r\nplatformio.ini - Project Configuration File\n\nError: Traceback (most recent call last):\r\n File \"c:\\users\\ishan\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 120, in main\r\n cli(None, None, None)\r\n File \"c:\\users\\ishan\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 700, in __call__\r\n return self.main(*args, **kwargs)\r\n File \"c:\\users\\ishan\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 680, in main\r\n rv = self.invoke(ctx)\r\n File \"c:\\users\\ishan\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1027, in invoke\r\n return _process_result(sub_ctx.command.invoke(sub_ctx))\r\n File \"c:\\users\\ishan\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 873, in invoke\r\n return ctx.invoke(self.callback, **ctx.params)\r\n File \"c:\\users\\ishan\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 508, in invoke\r\n return callback(*args, **kwargs)\r\n File \"c:\\users\\ishan\\.platformio\\penv\\lib\\site-packages\\click\\decorators.py\", line 16, in new_func\r\n return f(get_current_context(), *args, **kwargs)\r\n File \"c:\\users\\ishan\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\init.py\", line 92, in cli\r\n init_base_project(project_dir)\r\n File \"c:\\users\\ishan\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\init.py\", line 155, in init_base_project\r\n join(project_dir, \"platformio.ini\"))\r\n File \"C:\\Users\\Ishan\\.platformio\\python27\\Lib\\shutil.py\", line 83, in copyfile\r\n with open(dst, 'wb') as fdst:\r\nIOError: [Errno 2] No such file or directory: 'C:\\\\Users\\\\Ishan\\\\Documents\\\\PlatformIO\\\\Projects\\\\test\\\\platformio.ini'\r\n\r\n============================================================\r\n\r\nAn unexpected error occurred. Further steps:\r\n\r\n* Verify that you have the latest version of PlatformIO using\r\n `pip install -U platformio` command\r\n\r\n* Try to find answer in FAQ Troubleshooting section\r\n https://docs.platformio.org/page/faq.html\r\n\r\n* Report this problem to the developers\r\n https://github.com/platformio/platformio-core/issues\r\n\r\n============================================================"

IshanBhavsar, 2019-04-25:
Somehow Platform IO does not have file access it is not creating directories and files

ivankravets, 2019-05-03:
It seems that antivirus tools block PlatformIO. Please create a project in other custom directory.

closed 2019-05-03 by ivankravets

IshanBhavsar, 2019-05-05:
tried custom location still does not work