platformio/platformio-core #2411: Home: Could not initialize project

Lejacming, 2019-04-25:
PIO Core Call Error: "The next files/directories have been created in C:\\Users\\shaoy\\Documents\\PlatformIO\\Projects\\stm8test\r\ninclude - Put project header files here\r\nlib - Put here project specific (private) libraries\r\nsrc - Put project source files here\r\nplatformio.ini - Project Configuration File\r\nPlatformManager: Installing ststm8\r\nDownloading\r\nDownloading\r\nUnpacking\r\nUnpacking\r\nststm8 @ 1.0.1 has been successfully installed!\r\nPackageManager: Installing tool-stm8binutils @ 0.230.0\r\r\nDownloading\r\nDownloading\r\nUnpacking\r\nUnpacking\r\ntool-stm8binutils @ 0.230.0 has been successfully installed!\r\nPackageManager: Installing toolchain-sdcc @ ~1.30804.10766\r\r\nDownloading\r\nDownloading\r\nUnpacking\r\nUnpacking\r\ntoolchain-sdcc @ 1.30804.10766 has been successfully installed!\r\nThe platform 'ststm8' has been successfully installed!\r\r\nThe rest of packages will be installed automatically depending on your build environment.\n\nError: Processing stm8sdisco (platform: ststm8; board: stm8sdisco; framework: spl)\r\n\r\nPackageManager: Installing framework-ststm8spl @ 0.20301.181217\r\nDownloading\r\nDownloading\r\nUnpacking\r\nUnpacking\r\nVerbose mode can be enabled via `-v, --verbose` option\r\nCONFIGURATION: https://docs.platformio.org/page/boards/ststm8/stm8sdisco.html\r\nPLATFORM: ST STM8 > ST STM8S-DISCOVERY\r\nHARDWARE: STM8S105C6T6 16MHz 2KB RAM (32KB Flash)\r\nDEBUG: CURRENT(stlink) ON-BOARD(stlink)\r\nError: Could not parse library files for the target.\r\nstm8s.h:2723:25: fatal error: stm8s_conf.h: No such file or directory\r\n\r\n********************************************************************\r\n* Looking for stm8s_conf.h dependency? Check our library registry!\r\n*\r\n* CLI  > platformio lib search \"header:stm8s_conf.h\"\r\n* Web  > https://platformio.org/lib/search?query=header:stm8s_conf.h\r\n*\r\n********************************************************************\r\n\r\ncompilation terminated.\r\n [ERROR] Took 9.22 seconds"

Lejacming, 2019-04-25:
Kidding me?

ivankravets, 2019-05-03:
Duplicate of https://github.com/platformio/platform-ststm8/issues/6

closed 2019-05-03 by ivankravets