platformio/platformio-core #2437: Home: Could not import project

GitHubBeinner, 2019-04-30:
PIO Core Call Error: "The next files/directories have been created in /Users/didi/Documents/PlatformIO/Projects/190430-161043-kendryte210-standalone-sdk_hello\ninclude - Put project header files here\nlib - Put here project specific (private) libraries\nsrc - Put project source files here\nplatformio.ini - Project Configuration File\n\nError: Could not find a version that satisfies the requirement '~8.2.0' for your system 'darwin_x86_64'"

ivankravets, 2019-05-03:
There is no support for macOS for https://github.com/sipeed/platform-kendryte210

Please file a feature request here https://github.com/sipeed/platform-kendryte210/issues

ivankravets, 2019-05-03:
CC / @btx000

closed 2019-05-03 by ivankravets