platformio/platformio-core #2442: Error when using 99-platformio-udev.rules

definitio, 2019-05-01:
### Configuration

**Operating system**: Manjaro

**PlatformIO Version** (`platformio --version`):
3.6.5

### Description of problem
The warning `systemd-udevd[330]: Specified group 'plugdev' unknown` is shown in system log.

#### Steps to Reproduce

1. Install udev rules for PlatformIO.

### Actual Results
`GROUP="plugdev", TAG+="uaccess"` are used in rules.

### Expected Results
The correct action is to use systemd session mechanism that tags the devices as "uaccess" without "plugdev" group. `TAG+="udev-acl"` also can be added for compatibility with old OSes.

### Additional info
The "plugdev" group seems to be a special group from Debian/Ubuntu and is not generally used on GNU/Linux.

ivankravets, 2019-05-03:
Thanks for the report! Please re-test with the latest development version http://docs.platformio.org/en/latest/installation.html#development-version

Or, `pio upgrade --dev`.

closed 2019-05-03 by ivankravets

definitio, 2019-05-03:
Fixed. Thanks!

septatrix, 2021-06-23:
Are there plans to switch back to `TAG+="uaccess"` from `MODE:=0666`? The later is discouraged and insecure compared to the latter and due to the use of `:=` can also not be overwritten by any other configuration which one might want to perform. Whereas the former correctly handles all of this and respects currently logged in users etc...

ivankravets, 2021-06-23:
@septatrix please open a new issue

commit 48c1aeae (2019-05-03): Fix "systemd-udevd" warnings in 99-platformio-udev.rules // Resolve #2442
commit 2d423a49 (2019-05-03): Fix "systemd-udevd" warnings in 99-platformio-udev.rules // Resolve #2442