platformio/platformio-core #2457: Home: Could not initialize project

xukang15216856538, 2019-05-06:
PIO Core Call Error: "The next files/directories have been created in C:\\Users\\Administrator\\Documents\\PlatformIO\\Projects\\test\r\ninclude - Put project header files here\r\nlib - Put here project specific (private) libraries\r\nsrc - Put project source files here\r\nplatformio.ini - Project Configuration File\n\nError:"

ivankravets, 2019-05-06:
Could you provide more details?

antoniobreton, 2019-05-07:
I have the same issue, when I try to create a new project using the Nucleo F031K6 using mbed framework

ivankravets, 2019-05-07:
@antoniobreton could you share all errors from VSCode > Help > Toggle Developer Tools > Console?

antoniobreton, 2019-05-07:
Could not initialize project
PIO Core Call Error: "<Exception instance at 0x62084c8 with str error:\n Traceback (most recent call last):\n File \"C:\\.platformio\\packages\\contrib-pysite\\twisted\\python\\reflect.py\", line 448, in safe_str\n return str(o)\nUnicodeEncodeError: 'ascii' codec can't encode character u'\\xf3' in position 69: ordinal not in range(128)\n>

This is the error that platformio throws me

ivankravets, 2019-05-07:
What is your Windows OS version? Could you try to create a project in another folder which path does not contain non-ascii chars?

antoniobreton, 2019-05-08:
I have Windows 10, I'm going to try  to create the project in another folder

antoniobreton, 2019-05-08:
I created a folder directly in C:/ drive, When I created the project in that folder the problem was fixed. 

Thank you.

ivankravets, 2019-05-23:
Should be fixed in the latest PIO Core when Python 3 is used.

closed 2019-05-23 by ivankravets