platformio/platformio-core #2458: Cannot install library to project storage using GUI

kobuki, 2019-05-06:
As the title says. Installing eg. `id=2941`  (HTU21D by ejoyneering) from the command line like `platformio lib install 2941` works properly and installs the lib under .piolibdeps, I can update the index, use the lib, etc. However, the GUI doesn't offer to install the lib locally for the project. Image of the GUI attached.

Additional info: the project was moved from the default location in my Windows profile dir to my work directory on another drive. System is Win7 x64, VSCode 1.32.1, PIO Home 2.0.0, Core 3.6.5

[image]

ivankravets, 2019-05-23:
You need to open your project with PIO Home > OPen Project. After this step, you will see projects listed in this dropdown menu.

kobuki, 2019-05-23:
@ivankravets: how is your response related to my original issue?

ivankravets, 2019-05-23:
Could you explain in details about your issue?

kobuki, 2019-05-23:
Could you please read my first post? I've explained and illustrated it.

ivankravets, 2019-05-23:
Answeredhere => https://github.com/platformio/platformio-core/issues/2458#issuecomment-495331541

PIO Home does not see your project because you have not opened it yet via PIO Home.

closed 2019-05-23 by ivankravets

kobuki, 2019-05-23:
OK, so opening in PIO Home is different from opening via File/Open... or restarting VSC and letting it re-open by itself? I didn't know that.

ivankravets, 2019-05-23:
Yes, you are right.