platformio/platformio-core #2462: Build Issues

mars000, 2019-05-07:
I'm running latest version of VSC and PlatformIO on Windows 10.

Even though I use lib_ignore in my `platformio.ini` (see screenshot) I still get errors during linking phase. I don't understand why as I though the `lib_ignore `will tell the compiler not to include the listed files when Linking. Am I missing something ?

[image]

ivankravets, 2019-05-07:
> The correct value for this option is a library name (not folder name).

See docs file:///Volumes/SOFT/Projects/GitHub/platformio/origin/docs/_build/html/projectconf/section_env_library.html#lib-ignore

ivankravets, 2019-05-07:
> The correct value for this option is a library name (not folder name).

See docs http://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-ignore

closed 2019-05-07 by ivankravets

mars000, 2019-05-07:
I see but given this is my own created e.g. `#include <myfile.h>` file that is within the same directory as src; do I need to register my own `#includes` as a new library in PlatfromIO ? If so is a easy guide on how to do that ?

mars000, 2019-05-07:
think I worked it out by reading this reference :-)   [assuming I'm on right track] https://docs.platformio.org/en/latest/librarymanager/config.html#library-config