platformio/platformio-core #2473: Home: Could not load PIO Account information

paulshrimp, 2019-05-09:
PIO Core Call Error: "CorePackageManager: Installing tool-pioplus @ ^2.1.4\r\nDownloading\r\nDownloading\r\nUnpacking\r\nUnpacking\n\n[Error 5] Access is denied: 'C:\\\\Users\\\\rowan\\\\.platformio\\\\packages\\\\tool-pioplus\\\\pioplus.exe' \r\nPlease manually remove the file `C:\\Users\\rowan\\.platformio\\packages\\tool-pioplus\\pioplus.exe`\r\n[Error 5] Access is denied: 'C:\\\\Users\\\\rowan\\\\.platformio\\\\packages\\\\tool-pioplus' \r\nPlease manually remove the file `C:\\Users\\rowan\\.platformio\\packages\\tool-pioplus`\r\nError: Could not install 'tool-pioplus' with version requirements '^2.1.4' for your system 'windows_amd64'.\r\n\r\nPlease try this solution -> http://bit.ly/faq-package-manager"

ivankravets, 2019-05-11:
Please close PlatformIO IDE, remove `C:\\Users\\rowan\\.platformio\\packages\\tool-pioplus` folder and restart VSCode. Sorry for the issue.

closed 2019-05-11 by ivankravets

cheezemanrich, 2019-06-03:
On Linux Mint I tried but it doesn't work.
Close PlatformIO IDE.
Go to your 'user' folder, in the view menu select 'Show Hidden Files', open the '.platformio' folder, open the 'packages' folder, delete the 'tool-pioplus' folder.
Open PlatformIO IDE

Message;
PIO Core Call Error: "CorePackageManager: Installing tool-pioplus @ ^2.4.0\ntool-pioplus @ 2.5.0 has been successfully installed!\n\nTraceback (most recent call last):\n  File \"<string>\", line 1, in <module>\n  File \"<string>\", line 38, in <module>\n  File \"<string>\", line 39, in <module>\n  File \"<string>\", line 18, in load_module\n  File \"<string>\", line 25, in <module>\nImportError: cannot import name get_project_shared_dir"

ivankravets, 2019-06-03:
@cheezemanrich it seems that you use the development version of PIO Core 4.0. Please open PIO IDE Terminal and type `pio upgrade --dev`.

cheezemanrich, 2019-06-03:
Worked perfect. Thank you.