platformio/platformio-core #2491: Home: Libraries: Could not update library

witan4702, 2019-05-13:
PIO Core Call Error: "Library Storage: C:\\Users\\witan\\Documents\\Arduino\\libraries\r\nUpdating AutoConnect                     @ 0.9.7          [0.9.8]\r\r\nUninstalling AutoConnect @ 0.9.7: \t[OK]\r\nLooking for AutoConnect library in registry\r\nFound: https://platformio.org/lib/show/2678/AutoConnect\n\nError: Traceback (most recent call last):\r\n  File \"c:\\users\\witan\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 120, in main\r\n    cli(None, None, None)\r\n  File \"c:\\users\\witan\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 700, in __call__\r\n    return self.main(*args, **kwargs)\r\n  File \"c:\\users\\witan\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 680, in main\r\n    rv = self.invoke(ctx)\r\n  File \"c:\\users\\witan\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1027, in invoke\r\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\r\n  File \"c:\\users\\witan\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1027, in invoke\r\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\r\n  File \"c:\\users\\witan\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 873, in invoke\r\n    return ctx.invoke(self.callback, **ctx.para

ivankravets, 2019-05-13:
What is a version of PIO Core? Please open PlatformIO IDE Terminal and type `pio --version.`

witan4702, 2019-05-14:
PlatformIO, version 4.0.0a12

ivankravets, 2019-05-16:
I can't reproduce this issue. Could you try to run `pio update` from PlatformIO IDE Terminal and provide full output?

closed 2019-05-23 by ivankravets