platformio/platformio-core #2494: Floating point exception (core dumped)

Riyuzakii, 2019-05-13:
I'm trying to upload firmware to Odroid XU4, SmartPower 2.
I made a change to increase logging frequency of power consumption data.
The code for firmware was taken from here: [https://github.com/hardkernel/smartpower2.git](https://github.com/hardkernel/smartpower2.git)
These two commands run fine(with some warnings about some enumeration value and unused variables):
```
sudo platformio run
sudo platformio run --target upload
```
when I run the following command, I'm getting the shown error:

```
sudo platformio run --target uploadfs

Processing esp12e (platform: espressif8266; board: esp12e; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp12e.html
PLATFORM: Espressif 8266 > Espressif ESP8266 ESP-12E
HARDWARE: ESP8266 160MHz 80KB RAM (4MB Flash)
Converting smartpower2.ino
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 30 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <WebSockets> 2.1.3
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <SPI> 1.0
|   |-- <Ethernet(esp8266)> 1.0.4
|   |   |-- <SPI> 1.0
|-- <LiquidCrystal_I2C>
|   |-- <Wire> 1.0
|-- <mcp4652>
|   |-- <Wire> 1.0
|-- <Hash> 1.0
|-- <ESP8266WiFi> 1.0
|-- <SimpleTimer>
|-- <ESP8266WebServer> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <Wire> 1.0
Building SPIFFS image from 'data' directory to .pioenvs/esp12e/spiffs.bin
Floating point exception (core dumped)
*** [.pioenvs/esp12e/spiffs.bin] Error 136
====================================================================== [ERROR] Took 0.76 seconds ======================================================================

```
Can someone(@ivankravets ) help me out with this? There's another [issue](https://github.com/platformio/platformio-core/issues/469) like this but that issue doesn't really address the problem plus the error code is different.
Thanks

ivankravets, 2019-05-13:
Please forward your question to https://community.platformio.org/

closed 2019-05-13 by ivankravets

Riyuzakii, 2019-05-15:
Hi @ivankravets can please help out with this issue, I posted the issue on https://community.platformio.org/t/floating-point-exception-core-dumped/7797
but I haven't gotten any replies.