platformio/platformio-core #2531: Library Dependency Finder won't work on an unused class of a library, resulting on an uncompiling code

Alv3s, 2019-05-21:
#### Issue
Code doesn't compile when a dependency is inside of an unused header of a library. The issue seemes to be that, as the header of the unused file is not used, LDF does not parse it, resulting in a dependency that is never linked when compiling and thus resulting in a uncompiled code.

**PlatformIO Version**: 3.6.7

#### Steps to Reproduce

1. Create a Library "SomeLibrary" by creating a folder inside lib.
2. Create a class (a .h and .cpp file) called "SomeUnusedClass

## SomeUnusedClass.h
```
#pragma once

#include "Ticker.h"

class SomeUnusedClass {
public:
    SomeUnusedClass();
private:
    Ticker* ticker;
};
```

## SomeUnusedClass.cpp
```
#include "SomeUnusedClass.h"

SomeUnusedClass::SomeUnusedClass()
{
    
}
```

3. Create another class (just header is needed) called SomeUsedClass.
## SomeUsedClass.h
```
#pragma once

class SomeUsedClass
{

};
```

4. Include in main.cpp the file "SomeUsedClass.h"
## main.cpp
```
#include <Arduino.h>
#include "SomeUsedClass.h"

void setup() {
  // put your setup code here, to run once:
}

void loop() {
  // put your main code here, to run repeatedly:
}
```
### Actual Results
The compilation fails because "Ticker.h" was not found, although "Ticker.h" is placed in a .h file and it should be parsed.

### Expected Results
The compilation was expected to be successfull.

### Additional info
I found two workarounds to the problem:
1. Set LDF mode to be in mode "deep".
2. Removing "SomeUnusedClass.cpp" and add the code directly to "SomeUnusedClass.h" to turn it into a header only class also solves the problem. This is troublesome since not always the lib is mine.

ivankravets, 2019-05-23:
See http://docs.platformio.org/en/latest/projectconf/section_env_build.html#src-filter

PlatformIO builds automatically ALL source files from `src` folder. As result, SomeUnusedClass.cpp will also be built.

Alv3s, 2019-05-23:
@ivankravets As explained in my initial code, SomeUnusedClassed is not placed inside ```src``` folder, so the problem still remains.

ivankravets, 2019-05-23:
Please provide a simple project to reproduce this issue.

Alv3s, 2019-05-23:
@ivankravets As requested.
[Issue2531.zip](https://github.com/platformio/platformio-core/files/3214389/Issue2531.zip)

ivankravets, 2019-05-25:
You used header file from a library, this instructs PlatformIO Build System to build this library. As a result, it processed `SomeUnusedClass.cpp` file which depends on SomeUnusedClass.h.

If you need dynamically control which source files from a library should be built, see http://docs.platformio.org/en/latest/librarymanager/config.html#srcfilter

closed 2019-05-25 by ivankravets