platformio/platformio-core #2543: Home: Could not load PIO Account information

assassins377, 2019-05-23:
PIO Core Call Error: "CorePackageManager: Installing tool-pioplus @ ^2.1.4\r\nDownloading\r\nDownloading\r\nUnpacking\r\nUnpacking\n\n[Error 5] : 'C:\\\\Users\\\\Wordstar\\\\.platformio\\\\packages\\\\tool-pioplus\\\\pioplus.exe' \r\nPlease manually remove the file `C:\\Users\\Wordstar\\.platformio\\packages\\tool-pioplus\\pioplus.exe`\r\n[Error 5] : 'C:\\\\Users\\\\Wordstar\\\\.platformio\\\\packages\\\\tool-pioplus' \r\nPlease manually remove the file `C:\\Users\\Wordstar\\.platformio\\packages\\tool-pioplus`\r\nError: Could not install 'tool-pioplus' with version requirements '^2.1.4' for your system 'windows_amd64'.\r\n\r\nPlease try this solution -> http://bit.ly/faq-package-manager"

assassins377, 2019-05-23:
7777

ivankravets, 2019-05-23:
Duplicate of #2473

closed 2019-05-23 by ivankravets