platformio/platformio-core #2557: VSCode/Windows: Path of $PROJECT_DIR gets broken

cyberman54, 2019-05-25:
- [X] **PlatformIO Core**.
   

### Configuration

**Operating system**: Windows 10

**PlatformIO Version** (`platformio --version`): 4.0.0.a15

### Description of problem

On a Windows10 system with VSCode, a build flag in platformio.ini containing a path resolves this path with missing backslashes:

` DARDUINO_LMIC_PROJECT_CONFIG_H=$PROJECT_DIR/lmic_config.h`

results to

` ARDUINO_LMIC_PROJECT_CONFIG_H=c:UsersMyNameFooBar/lmic_config.h`

but no to

` ARDUINO_LMIC_PROJECT_CONFIG_H=c:\Users\MyName\FooBar/lmic_config.h`

as it should?

ivankravets, 2019-05-25:
Does relative path work? `-DARDUINO_LMIC_PROJECT_CONFIG_H=lmic_config.h`?

Could you try `-DARDUINO_LMIC_PROJECT_CONFIG_H=${PROJECT_DIR.replace("\", "/")}/lmic_config.h`?

cyberman54, 2019-05-25:
Relative path gives error when the library tries to load the file.
The other option does not compile.

cyberman54, 2019-05-26:
@ivankravets addendum: this problem arised after last pio upgrade to v4.0.0a15, because library directory structure changed (.piolibdeps -> .pio). This breaks all relative paths in platformio.ini, so would be better to have absolute paths.

cyberman54, 2019-05-26:
I close this, since i am not sure if this is a side effect from application.

closed 2019-05-26 by cyberman54

arrio464, 2023-06-24:
Sorry to bother people in this thread, but I think I need to say some thing to avoid latecomers wasting hours here...

For Windows user, just use `{$PROJECT_DIR.replace("/", "\")` instead of `$PROJECT_DIR`.

For example, use 
`upload_command = openocd -f interface/cmsis-dap.cfg -f target/stellaris.cfg -c "program {$PROJECT_DIR.replace("/", "\")}/.pio/build/lptm4c123gh6pm/firmware.elf verify reset exit"` 
instead of 
`upload_command = openocd -f interface/cmsis-dap.cfg -f target/stellaris.cfg -c "program $PROJECT_DIR/.pio/build/lptm4c123gh6pm/firmware.elf verify reset exit"`.

mattbaker-digital, 2024-08-20:
For `build_flags` etc., I used double quotes to expand the value. E.g. `-I"${PROJECT_DIR}"/rest/of/path`