platformio/platformio-core #2559: Package cache expires too fast

mcspr, 2019-05-26:
In platformio.ini
```
lib_deps =
    ArduinoJson@5.13.4
```

```
$ pio run
$ rm -r .pio/libdeps/<env>/ArduinoJson_ID64
$ ...wait some time...
$ pio run
```
If run fast enough, cache never expires. Otherwise, package is deleted and redownloaded after 1 minute.

Following maps expiration time shortcuts:
https://github.com/platformio/platformio-core/blob/67f7b6cda39b607e3c093bbb254443ef93cd3a68/platformio/app.py#L199

This is 1 minute instead of intended 30d?
https://github.com/platformio/platformio-core/blob/67f7b6cda39b607e3c093bbb254443ef93cd3a68/platformio/managers/package.py#L142

Shouldn't expiration time be "never" for these, since urls are unique per version?

_Originally posted by @mcspr in https://github.com/platformio/platformio-core/issues/1696#issuecomment-495916657_

ivankravets, 2019-05-27:
Thank you so much for the report! Please re-test with `pio upgrade --dev`.

closed 2019-05-27 by ivankravets

mcspr, 2019-05-27:
It works, thanks!

commit 3f96dc14 (2019-05-27): Fixed an issue when package cache (Library Manager) expires too fast // Resolve #2559