platformio/platformio-core #2567: Home: Could not load boards list

jacksunworld, 2019-05-27:
PIO Core Call Error: "Error: Traceback (most recent call last):\r\n  File \"c:\\users\\suncheng\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 120, in main\r\n    cli(None, None, None)\r\n  File \"c:\\users\\suncheng\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 700, in __call__\r\n    return self.main(*args, **kwargs)\r\n  File \"c:\\users\\suncheng\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 680, in main\r\n    rv = self.invoke(ctx)\r\n  File \"c:\\users\\suncheng\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1027, in invoke\r\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\r\n  File \"c:\\users\\suncheng\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 873, in invoke\r\n    return ctx.invoke(self.callback, **ctx.params)\r\n  File \"c:\\users\\suncheng\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 508, in invoke\r\n    return callback(*args, **kwargs)\r\n  File \"c:\\users\\suncheng\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\boards.py\", line 29, in cli\r\n    return _print_boards_json(query, installed)\r\n  File \"c:\\users\\suncheng\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\boards.py\", line 81, in _print_boards_json\r\n    for board in _get_boards(installed):\r\n  File \"c:\\users\\suncheng\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\boards.py\", line 76, in _get_boards\r\n    return pm.get_installed_boards() if installed else pm.get_all_boards()\r\n  File \"c:\\users\\suncheng\\.platformio\\penv\\lib\\site-packages\\platformio\\managers\\platform.py\", line 182, in get_all_boards\r\n    know_boards = [\"%s:%s\" % (b['platform'], b['id']) for b in boards]\r\nUnicodeDecodeError: 'ascii' codec can't decode byte 0xb8 in position 15: ordinal not in range(128)\r\n\r\n============================================================\r\n\r\nAn unexpected error occurred. Further steps:\r\n\r\n* Verify that you have the latest version of PlatformIO using\r\n  `pip install -U platformio` command\r\n\r\n* Try to find answer in FAQ Troubleshooting section\r\n  https://docs.platformio.org/page/faq.html\r\n\r\n* Report this problem to the developers\r\n  https://github.com/platformio/platformio-core/issues\r\n\r\n============================================================"

ivankravets, 2019-05-27:
Did you install dev/platforms with boards that contain non-ascii chars in their names?

ivankravets, 2019-05-28:
Please reopen if you still need help

closed 2019-05-28 by ivankravets

jacksunworld, 2019-05-29:
i reinstanlll the platformio. and resolved the
problem. thanks for reply