platformio/platformio-core #2569: Can't  enable --specs=nano.specs

HamidSaffari, 2019-05-28:
Hi, and best regards to creators.
to enabling nano.specs for STM32DUINO as discussed here which cut the FLASH usage in half:
https://github.com/bblanchon/ArduinoJson/issues/393

"the solution seems to be this: add:
--specs=nano.specs
at the very end of line in Documents\Arduino\hardware\Arduino_STM32\STM32F1\platform.txt on the line that starts with "recipe.c.combine.pattern=...."
as suggested here:
http://www.stm32duino.com/viewtopic.php?f=28&t=1596&p=20642#p20627
"
now how could I enable it in PlatformIO? I tried to do what they said in:
C:\Users\[user]\.platformio\packages\framework-arduinoststm32-maple\STM32F1\platform.txt
but it does not work.

HamidSaffari, 2019-05-28:
I find the solution: we should use separate python scripts to do that so add "extra_scripts = extra_script.py" in the platform.ini and then create extra_script.py file and write this to it:

Import("env")
env.Append(
  LINKFLAGS=[	"--specs=nano.specs" ] )

salute to you programmers that put this many helpful descriptions for your golden project.

ivankravets, 2019-05-28:
> I find the solution: we should use separate python scripts to do that so add "extra_scripts = extra_script.py"

This is the right temporary workaround. There is an issue for this case https://github.com/platformio/platformio-core/issues/2144 . Please subscribe to it.

closed 2019-05-28 by ivankravets