platformio/platformio-core #2572: Dependency resolver choses bogus EspSoftwareSerial 5.1.0 which shouldn't exist

andig, 2019-05-28:
- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

Consider this code:

    #include <SoftwareSerial.h>           // Using GPIOs for Serial Modbus communication
    SoftwareSerial sSerial(D1, D2); //SoftwareSerial object (RX, TX)

With this `platformio.ini`

    [env:esp8266]
    platform = espressif8266
    board = d1_mini_pro
    framework = arduino

    lib_deps=
    ..
    EspSoftwareSerial@5.0.4

    Dependency Graph
    ..
    |-- <EspSoftwareSerial> 5.0.4 #25e6e99

compiles fine.

However, when I specify a wider library version:

    lib_deps=
    ..
    EspSoftwareSerial@^5.0.4

Then the compile fails:

    Dependency Graph
    ..
    |-- <EspSoftwareSerial> 5.1.0

    /Users/andig/Documents/PlatformIO/Projects/190528-093927-esp01_1m/src/SimpleEVSE-WiFi.ino:90:30: error: no matching function for call to 'SoftwareSerial::SoftwareSerial(const uint8_t&, const uint8_t&)'

In `.piolibdeps/EspSoftwareSerial_ID168/library.json` it shows `5.1.0`.

The EspSoftwareSerial@5.1.0 that the dependency resolver has found does not exist in the source repo at https://github.com/plerup/espsoftwareserial. 
PIO home shows https://github.com/plerup/espsoftwareserial/commit/25e6e99e7257a0d094aabe6acec1014a212ac7e6 as installed version which is the latest commit. Even that commit has the correct initalizer, however no version has been released.

So where does that bogus 5.1.0 version come from and why does the resolver go up to 5.1.* when the requirement is ^5.0.4 (which should limit it to <5.1.0 as I understand SemVer)?

ivankravets, 2019-06-05:
Please open PlatformIO IDE Terminal and type
```
pio upgrade --dev
pio update
```

Restart VSCode. Does it work now?

andig, 2019-06-09:
In VScode, still old PIO is used after upgrade:

```
Obsolete PIO Core v3.6.7 is used (previous was 4.0.0a19)
Please remove multiple PIO Cores from a system:
https://docs.platformio.org/page/faq.html#multiple-pio-cores-in-a-system
```

https://docs.platformio.org/en/latest/faq.html#multiple-pio-cores-in-a-system only explains about atom. How would I upgrade the core inside vscode?

ivankravets, 2019-06-10:
Documentation has been updated. See https://docs.platformio.org/en/latest/ide/vscode.html#settings and `platformio-ide.useBuiltinPIOCore`.

andig, 2019-06-10:
LGTM. `EspSoftwareSerial^5.0`correctly choses `5.0.4` now instead of bogus `5.1.0`.

closed 2019-06-10 by andig

ivankravets, 2019-06-10:
Great! Happy coding with PlatformIO! ;)