platformio/platformio-core #2573: Improve handling library.json dependencies

ivankravets, 2019-05-28:
Currently PlatformIO installs library.json dependencies only if a library was installed via `$ pio lib` CLI or specifically declared via `lib_deps`. It fails with `Error: Could not find `xxx` dependency for `yyy` library.

nomis, 2019-08-17:
I also have a similar related problem... when I try to override an automatically downloaded library (because I'm developing a newer version of it) by putting a newer version in my `lib/` folder it has no effect (on my project's dependencies) unless I delete the `.pio/` folder to remove the previously downloaded version. This applies even though `platformio.ini` and any relevant `library.json` specifies the newer version.

ivankravets, 2019-08-17:
Hm... The `lib` folder of a project should have higher priority than `libdeps`. Could you try to reproduce this issue? you can check with `--verbose` which library is actually used.

nomis, 2019-08-17:
It continues to use `uuid-log@2.0.0` even after I add `uuid-log@2.0.3` to `lib/`:
```
simon@rincewind ~/src/fridge master $ rm -rf .pio
simon@rincewind ~/src/fridge master $ ls -l lib/
total 4
drwxrwxr-x 7 simon simon 4096 Aug 17 17:07 uuid-console
simon@rincewind ~/src/fridge master $ platformio run -v
*************************************************************************************************************************************************************************************************************************************************************************************
Obsolete PIO Core v4.0.0 is used (previous was 4.0.1b3)
Please remove multiple PIO Cores from a system:
https://docs.platformio.org/page/faq.html#multiple-pio-cores-in-a-system
*************************************************************************************************************************************************************************************************************************************************************************************
Processing d1_mini (build_unflags: -fno-rtti; board_build.f_cpu: 160000000L; build_flags: -std=c++11 -Os -flto -Wall -Wextra -Werror, -DPIO_FRAMEWORK_ARDUINO_LWIP2_IPV6_LOW_MEMORY, -DARDUINOJSON_EMBEDDED_MODE=1, -DARDUINOJSON_ENABLE_STD_STRING=1, -DARDUINOJSON_ENABLE_STD_STREAM=0, -DARDUINOJSON_ENABLE_ARDUINO_STRING=0, -DARDUINOJSON_ENABLE_ARDUINO_STREAM=1, -DARDUINOJSON_ENABLE_ARDUINO_PRINT=1, -DARDUINOJSON_USE_DOUBLE=0, -DARDUINOJSON_USE_LONG_LONG=0, !echo "-DFRIDGE_REVISION=\\\""$(git describe --dirty=+ --always)"\\\"" -Wl,-Teagle.flash.4m1m.ld; upload_port: /dev/serial/by-path/pci-0000:00:14.0-usb-0:5:1.0-port0; lib_deps: uuid-common@^1.0.1, uuid-log@^2.0.3, uuid-console@^0.1.0, ArduinoJson@~6.11.4; monitor_speed: 115200; platform: espressif8266@2.2.2; upload_speed: 1000000; board: d1_mini; framework: arduino; lib_ldf_mode: off)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini.html
PLATFORM: Espressif 8266 2.2.2 > WeMos D1 R2 and mini
HARDWARE: ESP8266 160MHz, 80KB RAM, 4MB Flash
PACKAGES: toolchain-xtensa 2.40802.190218 (4.8.2), tool-esptool 1.413.0 (4.13), tool-esptoolpy 1.20600.0 (2.6.0), framework-arduinoespressif8266 2.20502.0 (2.5.2)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ soft
Looking for uuid-common library in registry
Found: https://platformio.org/lib/show/6624/uuid-common
LibraryManager: Installing id=6624 @ ^1.0.1
Using cache: /home/simon/.platformio/.cache/8a/c648d028a95ff5e82ff93cea112f848a
uuid-common @ 1.0.1 has been successfully installed!
Looking for uuid-log library in registry
Found: https://platformio.org/lib/show/6625/uuid-log
LibraryManager: Installing id=6625 @ ^2.0.3
Using cache: /home/simon/.platformio/.cache/38/5714e4f337bccc8a80ed1ed6cd316238
uuid-log @ 2.0.0 has been successfully installed!
Installing dependencies
LibraryManager: Installing id=6624 @ ^1.0.0
uuid-common @ 1.0.1 is already installed
Looking for ArduinoJson library in registry
Found: https://platformio.org/lib/show/64/ArduinoJson
LibraryManager: Installing id=64 @ ~6.11.4
Using cache: /home/simon/.platformio/.cache/93/0b832423832d5c744b9902f0a1f03993
ArduinoJson @ 6.11.4 has been successfully installed!
Found 32 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <uuid-common> 1.0.1 (/home/simon/src/fridge/.pio/libdeps/d1_mini/uuid-common_ID6624)
|-- <uuid-log> 2.0.0 (/home/simon/src/fridge/.pio/libdeps/d1_mini/uuid-log_ID6625)
|   |-- <uuid-common> 1.0.1 (/home/simon/src/fridge/.pio/libdeps/d1_mini/uuid-common_ID6624)
|-- <uuid-console> 0.1.0 (/home/simon/src/fridge/lib/uuid-console)
|   |-- <uuid-log> 2.0.0 (/home/simon/src/fridge/.pio/libdeps/d1_mini/uuid-log_ID6625)
|   |   |-- <uuid-common> 1.0.1 (/home/simon/src/fridge/.pio/libdeps/d1_mini/uuid-common_ID6624)
|   |-- <uuid-common> 1.0.1 (/home/simon/src/fridge/.pio/libdeps/d1_mini/uuid-common_ID6624)
|-- <ArduinoJson> 6.11.4 (/home/simon/src/fridge/.pio/libdeps/d1_mini/ArduinoJson_ID64)
...
simon@rincewind ~/src/fridge master $ ln -s ~/src/mcu-uuid-log lib/uuid-log
simon@rincewind ~/src/fridge master $ ls -l lib/
total 4
drwxrwxr-x 7 simon simon 4096 Aug 17 17:07 uuid-console
lrwxrwxrwx 1 simon simon   28 Aug 17 21:01 uuid-log -> /home/simon/src/mcu-uuid-log
simon@rincewind ~/src/fridge master $ platformio run -v
*************************************************************************************************************************************************************************************************************************************************************************************
Obsolete PIO Core v4.0.0 is used (previous was 4.0.1b3)
Please remove multiple PIO Cores from a system:
https://docs.platformio.org/page/faq.html#multiple-pio-cores-in-a-system
*************************************************************************************************************************************************************************************************************************************************************************************
[... cut ...]
LDF Modes: Finder ~ off, Compatibility ~ soft
Found 33 compatible libraries
[... 87 lines cut ...]
		"type": "git",
		"url": "https://github.com/nomis/mcu-uuid-console.git"
	},
	"version": "0.1.0",
	"license": "GPL-3.0-or-later",
	"export": {
		"exclude": [
			".travis.yml",
			"test/*"
		]
	},
	"frameworks": [
		"arduino"
	],
	"dependencies": {
		"uuid-common": "^1.0.1",
		"uuid-log": "^2.0.3"
	},
	"build": {
		"flags": "-Wall -Wextra",
		"libLDFMode": "off"
	}
}
simon@rincewind ~/src/fridge master $ 
```

nomis, 2019-08-17:
`2.0.0` isn't a valid version for `^2.0.3` either, because that implies `>=2.0.3,<3.0.0`, so it should really cause an error when `2.0.3` doesn't exist.

nomis, 2019-08-30:
If I remove the `@version` from the `lib_deps` then it uses the local version. There should probably be a warning about specifying a version when there's a local directory present too.

programminghoch10, 2020-09-27:
Hello,
I've noticed that dependencies of `/lib/somelib/library.json` do not get picked up by the dependency finder.
Is this related to this or should i open another issue?

ivankravets, 2020-09-27:
@programminghoch10 please open a new issue and provide a simple project to reproduce it. Thanks!

Javerre, 2020-10-12:
@programminghoch10  I think your issue is the same as #2910? That issue included steps to reproduce but was closed by @ivankravets referencing this issue. Can you confirm?

vortigont, 2020-10-18:
Is there any known workaround for this at least?

ivankravets, 2020-10-19:
> Is there any known workaround for this at least?

Yes, add dependencies to `lib_deps`:
- https://docs.platformio.org/en/latest/librarymanager/quickstart.html

vortigont, 2020-10-20:
@ivankravets thanks for the tip with _lib_deps_, it works indeed.
The sad part - it is not always works as expected. In some cases I've faced quite weird behaviour doing this way - the project builds and links fine (at least no errors) but being uploaded to the MCU it just crashes on boot. Do not want to go deep in details here but uploading exact the same library to the remote git server and specifying URL in platformio.ini for the project instantly fixes the issue. Might be some strange linking side effects as it seems to me.
Let me know if any details might be useful here.
Thanks!

Bascy, 2021-03-28:
I aslo want to use the /lib folder, so I can make use it to clone git submodules of my project.
But as these cloned submodules are not included in the Library Dependency Finder, they also get downloaded in .pio/libdeps

closed 2022-03-08 by ivankravets

Bascy, 2022-03-08:
Sorry, but it is still not working as I expected (or I'm expecting the wrong thing ...)

Created a git repo with a platformio project, installed a submodule which is downloaded in the `/lib` folder of the project
This submodule has several dependencies stated in the library,.json file

If I don't include `lib_deps = <submodule>` in the platformio, then none of the dependencies are downloaded in the `.pio/esp32/dev/lib_deps` folder.

If I do include  `lib_deps = <submodule>` then the submodule and all of its dependencies are downloaded in the `.pio/esp32/dev/lib_deps`

What I would expect is that folders in `/lib` are seen as already present by the dependency finder. The `library.json` files in the folders in `/lib` should be included in the dependency finder but dependencies that are already present in the `/lib` folder should not downloaded be again in the `lib_deps` folder

Are my expectations correct?

nomis, 2022-03-08:
This is still not working with version 5.3.0a3:

## Scenario 1 (test using current project)
Using https://github.com/nomis/mcu-uuid-console/

1. Run `rm -rf test/build/example-DigitalIO/.pio`
2. Run `platformio run -d test/build/example-DigitalIO`

It finds the `lib/uuid-console` library (which is this project itself) but ignores its dependencies so the build fails.

## Scenario 2 (override library with local version)
Using https://github.com/nomis/scd30/

1. Put a symlink to `mcu-uuid-console` as `lib/uuid-console`
2. Increment the version of `lib/mcu-uuid-console/library.json` to 0.8.1 (so that it is visibly different)
3. Run `rm -rf .pio`
4. Run `platformio run`

Version 0.8.0 is used instead of 0.8.1 in the `lib` directory.

This one is complicated by the fact that libraries are now namespaced by username in the PlatformIO registry but not in `library.json`. Even if I change the name in `library.json` to `nomis/uuid-console` it still doesn't use the local version.

ivankravets, 2022-03-09:
This is a separate issue. You mean dependencies declared in the private libs located in the `lib` folder.

@nomis could I ask you to create a new issue and copy this comment https://github.com/platformio/platformio-core/issues/2573#issuecomment-1062238583 ? 

Thanks!

nomis, 2022-03-09:
Which one (or both) are separate issues?

There is already an issue for scenario 2 that you closed: #2910

ivankravets, 2022-03-09:
Thanks! I've just reopened https://github.com/platformio/platformio-core/issues/2910

ivankravets, 2022-03-31:
> Automatically install dependencies of the local (private) libraries

Please re-test https://github.com/platformio/platformio-core/issues/2910

commit f28651ea (2022-03-08): Ensure package dependencies are installed // Resolve #2573