platformio/platformio-core #2574: Linking Arduino sketch fails with undefined reference to `setup'

andig, 2019-05-28:
What kind of issue is this?

- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

### Configuration

**Operating system**:

    OSX Darwin

**PlatformIO Version** (`platformio --version`):

    PlatformIO, version 3.6.7

### Description of problem

It seems that https://community.platformio.org/t/cli-ci-build-fails-on-arduino-project/486/3 is back in some form. 

I have a more complex `.ino` file in the root folder with `platformio.ini` configured for Arduino:

    ; Please visit documentation for the other options and examples
    ; https://docs.platformio.org/page/projectconf.html

    [platformio]
    env_default=esp8266

    [env:esp8266]
    platform = espressif8266
    #board = esp01_1m
    board = d1_mini_pro
    framework = arduino
    lib_deps=
    ArduinoJson@^5.1
    ESP Async WebServer@^1.2
    ESPAsyncUDP
    EspSoftwareSerial@5.0.4
    Hash
    MFRC522@^1.4
    ModbusMaster@^2.0
    SPI
    Time@^1.5

The final linking step fails:

    Processing esp8266 (platform: espressif8266; board: d1_mini_pro; framework: arduino)
    ---------------------------------------------------------------------------------------------------------
    Verbose mode can be enabled via `-v, --verbose` option
    CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini_pro.html
    PLATFORM: Espressif 8266 > WeMos D1 mini Pro
    HARDWARE: ESP8266 80MHz 80KB RAM (16MB Flash)
    Library Dependency Finder -> http://bit.ly/configure-pio-ldf
    LDF MODES: FINDER(chain) COMPATIBILITY(soft)
    Collected 44 compatible libraries
    Scanning dependencies...
    Dependency Graph
    |-- <ArduinoJson> 5.13.4
    |-- <ESP Async WebServer> 1.2.0
    |   |-- <ESPAsyncTCP> 1.2.0
    |   |   |-- <ESP8266WiFi> 1.0
    |   |-- <ArduinoJson> 5.13.4
    |   |-- <Hash> 1.0
    |   |-- <ESP8266WiFi> 1.0
    |-- <ESPAsyncUDP>
    |-- <EspSoftwareSerial> 5.0.4
    |-- <Hash> 1.0
    |-- <MFRC522> 1.4.4
    |   |-- <SPI> 1.0
    |-- <ModbusMaster> 2.0.1
    |-- <SPI> 1.0
    |-- <Time> 1.5
    |-- <ESP8266WiFi> 1.0
    Linking .pioenvs/esp8266/firmware.elf
    .pioenvs/esp8266/libFrameworkArduino.a(core_esp8266_main.cpp.o):(.text._ZL12loop_wrapperv+0x4): undefined reference to `setup'
    .pioenvs/esp8266/libFrameworkArduino.a(core_esp8266_main.cpp.o):(.text._ZL12loop_wrapperv+0x8): undefined reference to `loop'
    .pioenvs/esp8266/libFrameworkArduino.a(core_esp8266_main.cpp.o): In function `loop_wrapper()':
    core_esp8266_main.cpp:(.text._ZL12loop_wrapperv+0x21): undefined reference to `setup'
    core_esp8266_main.cpp:(.text._ZL12loop_wrapperv+0x2d): undefined reference to `loop'
    collect2: error: ld returned 1 exit status
    *** [.pioenvs/esp8266/firmware.elf] Error 1

`loop` and `setup` are defined in the ino source like this:

````cpp
void ICACHE_RAM_ATTR setup() {
  // ...
}
void ICACHE_RAM_ATTR loop() {
  // ...
}
````

andig, 2019-05-28:
Here is an entire build log form travis: https://travis-ci.org/andig/vzero/builds/538177855

ivankravets, 2019-05-28:
Does it work if you convert INO to CPP? http://docs.platformio.org/en/latest/faq.html#convert-arduino-file-to-c-manually

andig, 2019-05-28:
I've tried apart from the function declaration (as the source is not mine and already pretty complex). Did not help.

ivankravets, 2019-05-28:
> I have a more complex .ino file in the root folder with platformio.ini configured for Arduino:

It should be in `src` folder, not in the root

andig, 2019-05-28:
I've added

    [platformio]
    src_dir=.

but doesn't help (I don't control the sources unfortunately and owner will want to keep it compatible with Arduino, whatever that means).

Even if wrong `src_dir` is the root cause- wouldn't it be good to issue a warning if the expected source dir doesn't exist at all?

ivankravets, 2019-05-28:
I need a simple project to reproduce this issue.

andig, 2019-05-28:
I could setup a repro with a few files, not too complex. Will update asap.

andig, 2019-05-28:
Ok, just did another clean build, working now. Whatever. I've noticed that the PIO builds are not very stable. Often something breaks on first run but is happy on retry. If thats worth an issue please let me know.

Otherwise- do you want to keep this open for the `src_dir` warning?

ivankravets, 2019-05-30:
Please provide a simple project to reproduce this issue.

andig, 2019-05-31:
Solved by setting src_dir=.

closed 2019-05-31 by andig

dontsovcmc, 2019-12-03:
I also has this error when call `platformio run --target upload` in the folder if only one file 'Basic.ino' or 'Basic.cpp' in the folder.