platformio/platformio-core #2578: Home: Could not initialize project

agambier, 2019-05-29:
Creating a new project for ESP32-CAM AI Thinker does not work.
IT reported me the following error message.

PIO Core Call Error: "The next files/directories have been created in C:\\Users\\agambier\\Desktop\\cam\\cam\r\ninclude - Put project header files here\r\nlib - Put here project specific (private) libraries\r\nsrc - Put project source files here\r\nplatformio.ini - Project Configuration File\n\nError: Processing esp32cam (platform: espressif32; board: esp32cam; framework: arduino)\r\n\r\nVerbose mode can be enabled via `-v, --verbose` option\r\nError: Unknown board ID 'esp32cam'\r\n [ERROR] Took 2.23 seconds"

ivankravets, 2019-05-29:
PIO IDE > PIO Home > Platforms > Updates. Restart PIO Home.

closed 2019-05-29 by ivankravets

agambier, 2019-05-29:
If it's not a issue... you should display a message box which explain that an update should be done.
Even if the board was selected in the board list when creating the project.