platformio/platformio-core #2581: Fatal error: SPI.h: No such file or directory

nirkons, 2019-05-29:
What kind of issue is this?
- [x ] **PlatformIO Core**.

### Configuration
**Operating system**: ESPHOME
espressif32@1.6.0;

### Description of problem
Adding Adafruit's new "Adafruit BusIO" library to ESPhome results in

> In file included from /data/pod5/.piolibdeps/Adafruit BusIO_ID6214/Adafruit_SPIDevice.cpp:1:0:
> /data/pod5/.piolibdeps/Adafruit BusIO_ID6214/Adafruit_SPIDevice.h:1:17: fatal error: SPI.h: No such file or directory

ivankravets, 2019-05-29:
Just add to some of your files in `src` folder `#include <SPI.h>`

closed 2019-05-29 by ivankravets

nirkons, 2019-05-29:
They do in fact have that.

pczekalski, 2020-05-21:
Same error with ESP8266 framework:
Adafruit BusIO_ID6214/Adafruit_SPIDevice.h:1:17: fatal error: SPI.h: No such file or directory

sprimeu, 2020-05-26:
Well as @ivankravets said, I could fix this issue by #including <SPI.h> in one of the src files. I guess PlatformIO reads through src files, to get an overall view of all the dependencies, to ensure, for instance, their header files are well specified in compiler's header directories.