platformio/platformio-core #2583: util.load_project_config() no longer working

cyberman54, 2019-05-29:
@ivankravets 
In `platformio.org` I'm calling a pre:build.py script which uses

```
# get platformio environment variables
project_config = util.load_project_config()
```

It seems commit #3df0140 broke this. How can i fix it?

ivankravets, 2019-05-29:
See updated example http://docs.platformio.org/en/latest/projectconf/advanced_scripting.html#custom-options-in-platformio-ini

We recommend using native Python config parser.

cyberman54, 2019-05-29:
@ivankravets 
But some .py modules of platformio-core still use the old method, e.g.

https://github.com/platformio/platformio-core/blob/develop/platformio/builder/main.py

ivankravets, 2019-05-29:
> But some .py modules of platformio-core still use the old method, e.g.

Which old methods?

cyberman54, 2019-05-29:
https://github.com/platformio/platformio-core/search?q=import+util&unscoped_q=import+util

cyberman54, 2019-05-29:
@ivankravets 

> TypeError: 'module' object is not callable:
>   File "C:\Users\MyUser\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 178:
>     env.SConscript(item, exports="env")

ivankravets, 2019-05-29:
We'll introduce totally new project option management API today/tomorrow. Still in testing. There will be opened API for build scripts:
```
env.GetProjectOption(option, default=None)
```

> from platformio import util

What is wrong with this import? I mean that `load_project_config` was totally removed from PlatformIO Core 4.0 because there is no single config `platformio.ini` which we had in PlatformIO Core. Now, you can pass a custom project config.

ivankravets, 2019-05-30:
Please reopen if you still need help

closed 2019-05-30 by ivankravets

irwanayeah, 2019-07-16:
yes, I have same issue and solved by http://docs.platformio.org/en/latest/projectconf/advanced_scripting.html#custom-options-in-platformio-ini
thx guys.