platformio/platformio-core #2594: Improve platform update error management through GUI

Navis-Raven, 2019-05-31:
What kind of issue is this?

- [X] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [X] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

The problem is that from time to time there is some trouble or errors while updating a platform through GUI here:
[image]

However, there is no error message nor anything we only see that it is still trying/loading for updating the platform. 
Otherwise if we do platformio updates all platform through scripts or command then we see the problem or the error message, and what we have to fix. 

So my suggestion is to improve error management through the GUI.

ivankravets, 2019-08-25:
Could you provide more details? I've just raised some exception in the update process and it works:

# CLI

[image]

# PIO Home

[image]

ivankravets, 2019-08-30:
Please reopen if you still need help

closed 2019-08-30 by ivankravets

Navis-Raven, 2019-09-09:
Sorry for the delay, I was in vacations. 

I think you didn''t understand. The problem occurs when you update through the GUI. 

Look at the screenshots you have, you have also some errors !
[image]

ivankravets, 2019-09-09:
Hi! This is normal. If you click on "Report problem", what do you see? Could you post report here? Thanks!

Navis-Raven, 2019-09-09:
That's ok, next time I'll see the report I'll post it here.