platformio/platformio-core #2597: Build process wont differantiate env and install on both

exocode, 2019-05-31:
What kind of issue is this?

- [X] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform
- [X] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

### Configuration

**Operating system**:
MacOS 10.13.6 (17G4015)

**PlatformIO Version** (`platformio --version`):
PlatformIO, version 3.6.7

### Description of problem

I wanna use `analogWrite` on a ESP32 as well on a ESP8266. ESP32 needs an additional lib, because it's (still) not implemented yet.

#### Steps to Reproduce

1. Setup new project with `platformio.ini`
2. Copy/Paste this into your `main.cpp
3. Compile
4. you will also see, that ESP32 AnalogWrite is present in heltec_wifi_kit_8 and .piolibdeps. 

### Actual Results

ESP8266 is not compiling, because **it's implementing ESP32 AnalogWrite, which is NOT in the ESP8266 env**.

### Expected Results

ESP32 AND ESP8266 should compile without `ESP32 AnalogRead`

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env]
monitor_speed = 115200

[env:heltec_wifi_kit_8]
platform = espressif8266
board = heltec_wifi_kit_8
framework = arduino
build_flags = -DESP8266

[env:heltec_wifi_kit_32]
platform = espressif32
board = heltec_wifi_kit_32
framework = arduino
lib_deps = ESP32 AnalogWrite
build_flags = -DESP32
```

**Source file to reproduce issue:**
```cpp
#include <Arduino.h>
#ifdef ESP32
    #include <analogWrite.h>  // <<<<---- SHOULD ONLY BE INCLUDED ON ESP32
#endif // ESP32

int brightStep = 1;
int brightness = 0;

void setup()
{
}

void loop()
{
    brightness += brightStep;
    if (brightness == 0 || brightness == 255)
    {
        brightStep = -brightStep;
    }

    analogWrite(LED_BUILTIN, brightness);
    delay(10);
}
```

### Additional info
**Here is a list of libs automatically implemented on build time: As you can see heltec_wifi_kit_8 also includes `ESP32 AnalogWrite` which should not be there:** 

[image]

ivankravets, 2019-05-31:
Resolved in PlatformIO Core 4.0. Please re-test with http://docs.platformio.org/en/latest/installation.html#development-version

closed 2019-05-31 by ivankravets

exocode, 2019-06-01:
Hm. It wont uninstall 

```
 janjezek   …/Desktop/DEMO-ESP-ARD-BLINK  pip uninstall platformio                                        2 ↵

DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7.
Skipping platformio as it is not installed.

 janjezek   …/Desktop/DEMO-ESP-ARD-BLINK  which pio                                                         ✔
/Users/janjezek/.platformio/penv/bin/pio

```

exocode, 2019-06-01:
I just deleted now `~/.platformio`... and installed via `pip install -U https://github.com/platformio/platformio-core/archive/develop.zip`

`pio` is now in 

```
 janjezek   …/Desktop/DEMO-ESP-ARD-BLINK  which pio                                                         ✔
/usr/local/bin/pio

 janjezek   …/Desktop/DEMO-ESP-ARD-BLINK  pio --version                                                     ✔
PlatformIO, version 4.0.0a16

 janjezek   …/Desktop/DEMO-ESP-ARD-BLINK  which pio                                                         ✔
/Users/janjezek/.platformio/penv/bin/pio

```

exocode, 2019-06-01:
Hm.. after installing I have still 3.6.7 in my VSCODE: I also tried to Reinstall the VSCode Plugin

[image]

exocode, 2019-06-01:
I found a checkbox in the VSCode settings: 

https://docs.platformio.org/en/latest/ide/vscode.html#platformio-ide-usedevelopmentpiocore

Which looks promising :-)
[image]
[image]

ivankravets, 2019-06-01:
Does it work now?

exocode, 2019-06-01:
@ivankravets: YES. I only had to fix my customPATH in the settings.

ivankravets, 2019-06-01:
You do not need to fix custom path. Be sure that there are no multiple PIO Cores in a system. After that, please remove `~/.platformio/penv` and restart VSCode. If you enbaled dev/version in IDE before, installer with install the latest dev/ version autoamtically.

Please re-test.