platformio/platformio-core #2606: Home: Could not initialize project

MCUdude, 2019-06-02:
I got this error message when trying to initialize a PIO project for MiniCore ATmega328P.

PIO Core Call Error: "The next files/directories have been created in /Users/hans/Documents/PlatformIO/Projects/Test\ninclude - Put project header files here\nlib - Put here project specific (private) libraries\nsrc - Put project source files here\nplatformio.ini - Project Configuration File\n\nError: Unknown board ID 'ATMEGA328P'"

ivankravets, 2019-06-02:
Could you provide an output from `pio platform list` command? Also, from `pio platform update --only-check`.

MCUdude, 2019-06-02:
The boards search is also kind of messed up.

[image]

MCUdude, 2019-06-02:
Sure!

```bash
$ pio platform list
atmelavr ~ Atmel AVR
====================
Atmel AVR 8- and 32-bit MCUs deliver a unique combination of performance, power efficiency and design flexibility. Optimized to speed time to market-and easily adapt to new ones-they are based on the industrys most code-efficient architecture for C and assembly programming.

Home: http://platformio.org/platforms/atmelavr
Frameworks: arduino, simba
Packages: toolchain-atmelavr, framework-arduinoavr, framework-simba, tool-avrdude, tool-micronucleus
Version: 1.13.0

atmelavr ~ Atmel AVR
====================
Atmel AVR 8- and 32-bit MCUs deliver a unique combination of performance, power efficiency and design flexibility. Optimized to speed time to market-and easily adapt to new ones-they are based on the industrys most code-efficient architecture for C and assembly programming.

Home: http://platformio.org/platforms/atmelavr
Frameworks: arduino, simba
Packages: toolchain-atmelavr, framework-arduinoavr, framework-simba, tool-avrdude, tool-micronucleus
Version: c7d34dd
```

``` bash
$ pio platform update --only-check
Platform Atmel AVR
--------
Checking atmelavr                        @ 1.13.0         [Up-to-date]
Checking toolchain-atmelavr              @ 1.50400.0      [Up-to-date]
Checking framework-arduinoavr            @ 3.10623.190424 [Up-to-date]
Checking tool-avrdude                    @ 1.60300.190424 [Up-to-date]
Checking tool-micronucleus               @ 1.200.4        [Up-to-date]

Platform Atmel AVR
--------
Checking atmelavr                        @ c7d34dd        [Up-to-date]
Checking toolchain-atmelavr              @ 1.50400.0      [Up-to-date]
Checking framework-arduinoavr            @ 4.0.0          [Up-to-date]
Checking tool-avrdude                    @ 1.60300.190424 [Up-to-date]
Checking tool-micronucleus               @ 1.200.4        [Up-to-date]

```

ivankravets, 2019-06-03:
Please re-run `pio update`, reopen PIO Home, everything should work now.  

I released a new 1.14.0 version https://github.com/platformio/platform-atmelavr/releases/tag/v1.14.0

Thanks!

closed 2019-06-03 by ivankravets