platformio/platformio-core #2610: Home: Could not import project

vivek6429, 2019-06-03:
PIO Core Call Error: "The next files/directories have been created in C:\\Users\\vi64\\Documents\\PlatformIO\\Projects\\190603-072144-esp8266-nonos-sdk-blink\r\ninclude - Put project header files here\r\nlib - Put here project specific (private) libraries\r\nsrc - Put project source files here\r\nplatformio.ini - Project Configuration File\n\nError: Could not install 'toolchain-xtensa' with version requirements '~1.40802.0' for your system 'windows_amd64'.\r\n\r\nPlease try this solution -> http://bit.ly/faq-package-manager"

ivankravets, 2019-06-03:
> Please try this solution -> http://bit.ly/faq-package-manager"

It seems that antivirus tools block PlatformIO.

closed 2019-06-03 by ivankravets

vivek6429, 2019-06-21:
Thank you for your response , i am new to both programming and iot .