platformio/platformio-core #2612: OpenOCD on RaspberryPi broken

ludetje, 2019-06-03:
What kind of issue is this?

- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

### Configuration

**Operating system**: Raspbian GNU/Linux 9 (stretch)

**PlatformIO Version** (`platformio --version`): PlatformIO, version 3.6.7

### Description of problem
The installation of the OpenOCD-Tool (tool-openocd @ 2.1000.190420) seems to be broken and compared to my Desktop (Linux Mint 19) a little older (tool-openocd @ 2.1000.190422). PlatformIO Versions are the same.

On the RaspberryPi OpenOCD fails to open a Connection to my STM32 NucleoF411RE Board due to missing Board-Config Files. Actually the Files are present, but not in the directory that OpenOCD looks for them.

#### Steps to Reproduce

1. Install PlatformIO
2. Connect Nucleo-Board to RaspberryPi
3. Build and Upload

4. Run OpenOCD standalone from inside .platformio/packages/tool-openocd/: `bin/openocd -d -f scripts/board/st_nucleo_f4.cfg`

### Actual Results
After Step 3 OpenOCD Fails with following output:
```
Open On-Chip Debugger 0.10.0+dev-00796-ga4ac561 (2019-04-20-17:09)
Licensed under GNU GPL v2
For bug reports, read
http://openocd.org/doc/doxygen/bugs.html
/home/pi/.platformio/packages/tool-openocd/scripts/board/st_nucleo_f4.cfg:7: Error: Can't find interface/stlink.cfg
in procedure 'script'
at file "embedded:startup.tcl", line 56
at file "/home/pi/.platformio/packages/tool-openocd/scripts/board/st_nucleo_f4.cfg", line 7
*** [upload] Error 1
```

If OpenOCD is called standalone with any given Board.cfg File in Verbose-Mode like in Step 4 it prints out the following lines:
```
Open On-Chip Debugger 0.10.0+dev-00796-ga4ac561 (2019-04-20-17:09)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
User : 13 13 command.c:544 command_print(): debug_level: 3
Debug: 14 13 options.c:184 add_default_dirs(): bindir=/tmp/pio-openocd/bin
Debug: 15 13 options.c:185 add_default_dirs(): pkgdatadir=/tmp/pio-openocd/share/openocd
Debug: 16 13 options.c:186 add_default_dirs(): exepath=/home/pi/.platformio/packages/tool-openocd/bin
Debug: 17 13 options.c:187 add_default_dirs(): bin2data=../share/openocd
Debug: 18 13 configuration.c:42 add_script_search_dir(): adding /home/pi/.openocd
Debug: 19 13 configuration.c:42 add_script_search_dir(): adding /home/pi/.platformio/packages/tool-openocd/bin/../share/openocd/site
Debug: 20 13 configuration.c:42 add_script_search_dir(): adding /home/pi/.platformio/packages/tool-openocd/bin/../share/openocd/scripts
Debug: 21 14 configuration.c:97 find_file(): found scripts/board/st_nucleo_f4.cfg
User : 22 15 command.c:715 command_run_line(): scripts/board/st_nucleo_f4.cfg:7: Error: Can't find interface/stlink.cfg
in procedure 'script' 
at file "embedded:startup.tcl", line 56
at file "scripts/board/st_nucleo_f4.cfg", line 7
```

### Expected Results
Expected result was that the firmware is beeing uploaded without errors.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:nucleo_f411re]
platform = ststm32
board = nucleo_f411re
framework = arduino
```

**Source file to reproduce issue:**
```cpp
#include <Arduino.h>

void setup()
{
    pinMode(LED_BUILTIN, OUTPUT);
}
void loop()
{
    digitalToggle(LED_BUILTIN);
    delay(200);
}
```

### Additional info
All installed OpenOCD-Scripts can be found under `.platformio/packages/tool-openocd/scripts/`
Only OpenOCD isn't configured to look for them there as you can see from Debug-Output 18 to 20.

I also have a workaround running for me. I added the directory `.platformio/packages/tool-openocd/share/openocd` and created a Sym-Link in it to the scripts-directory:
```
.platformio/packages/tool-openocd/share/openocd/ $ ls -la
total 8
drwxr-xr-x 2 pi pi 4096 May 29 16:53 .
drwxr-xr-x 5 pi pi 4096 Jun  3 10:56 ..
lrwxrwxr-x 1 pi pi   14 May 29 16:53 scripts -> ../../scripts/
```
With this Link the Script-Search in Debug-Line 20 does not fail and all Configurtions are loaded correctly and the firmware is uploaded.

ludetje, 2019-06-03:
### Edit
I already did a fresh install of PlatformIO and all Packages. It did not solve the problem.

ivankravets, 2019-06-10:
Please re-run `pio update`. Should work now. Thanks for the report!

ludetje, 2019-06-12:
Hmm... on my machine it still seems to be broken. When i run the follwoing command:
`~/.platformio/packages/tool-openocd/bin/openocd -d -f ~/.platformio/packages/tool-openocd/scripts/board/st_nucleo_f4.cfg`
i get the follwing output:
```
$ ~/.platformio/packages/tool-openocd/bin/openocd -d -f ~/.platformio/packages/tool-openocd/scripts/board/st_nucleo_f4.cfg
Open On-Chip Debugger 0.10.0+dev-00892-g159f11f (2019-06-10-12:25)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
User : 13 4 options.c:60 configuration_output_handler(): debug_level: 3
User : 14 4 options.c:60 configuration_output_handler(): 
Debug: 15 4 options.c:184 add_default_dirs(): bindir=/tmp/platformio/openocd/bin
Debug: 16 4 options.c:185 add_default_dirs(): pkgdatadir=/tmp/platformio/openocd
Debug: 17 4 options.c:186 add_default_dirs(): exepath=/home/pi/.platformio/packages/tool-openocd/bin
Debug: 18 4 options.c:187 add_default_dirs(): bin2data=../../openocd
Debug: 19 4 configuration.c:42 add_script_search_dir(): adding /home/pi/.openocd
Debug: 20 4 configuration.c:42 add_script_search_dir(): adding /home/pi/.platformio/packages/tool-openocd/bin/../../openocd/site
Debug: 21 4 configuration.c:42 add_script_search_dir(): adding /home/pi/.platformio/packages/tool-openocd/bin/../../openocd/scripts
User : 22 5 command.c:689 command_run_line(): embedded:startup.tcl:26: Error: Can't find /home/pi/.platformio/packages/tool-openocd/scripts/board/st_nucleo_f4.cfg
in procedure 'script' 
at file "embedded:startup.tcl", line 26
```
For me it looks like there is a typo in the script-search-path. In Debug 20 to 22 it seems to look in `.platformio/packages/openocd/scripts` for the scripts. But the Directory `openocd` should be called `tool-openocd` if i'm right? (Edit: maybe same goes for Debug 18)?

I removed my .platformio folder before and updated my udev-rules.

ivankravets, 2019-06-12:
Could you try?
```
$ ~/.platformio/packages/tool-openocd/bin/openocd -d -s ~/.platformio/packages/tool-openocd -f ~/.platformio/packages/tool-openocd/scripts/board/st_nucleo_f4.cfg
```

ludetje, 2019-06-12:
It outputs this:
```
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
User : 13 5 options.c:60 configuration_output_handler(): debug_level: 3
User : 14 5 options.c:60 configuration_output_handler(): 
Debug: 15 5 configuration.c:42 add_script_search_dir(): adding /home/pi/.platformio/packages/tool-openocd
Debug: 16 5 options.c:184 add_default_dirs(): bindir=/tmp/platformio/openocd/bin
Debug: 17 5 options.c:185 add_default_dirs(): pkgdatadir=/tmp/platformio/openocd
Debug: 18 5 options.c:186 add_default_dirs(): exepath=/home/pi/.platformio/packages/tool-openocd/bin
Debug: 19 5 options.c:187 add_default_dirs(): bin2data=../../openocd
Debug: 20 5 configuration.c:42 add_script_search_dir(): adding /home/pi/.openocd
Debug: 21 5 configuration.c:42 add_script_search_dir(): adding /home/pi/.platformio/packages/tool-openocd/bin/../../openocd/site
Debug: 22 5 configuration.c:42 add_script_search_dir(): adding /home/pi/.platformio/packages/tool-openocd/bin/../../openocd/scripts
User : 23 6 command.c:689 command_run_line(): embedded:startup.tcl:26: Error: Can't find /home/pi/.platformio/packages/tool-openocd/scripts/board/st_nucleo_f4.cfg
in procedure 'script' 
at file "embedded:startup.tcl", line 26
```

Actually i noticed something else. The Board Directory is now missing. :
```
~/.platformio/packages/tool-openocd/scripts $ ll
total 40
drwxr-xr-x 7 pi pi  4096 Jun 12 12:35 .
drwx------ 6 pi pi  4096 Jun 12 12:35 ..
-rw-r--r-- 1 pi pi  1601 Jun 10 16:00 bitsbytes.tcl
drwxr-xr-x 2 pi pi  4096 Jun 12 12:35 cpld
drwxr-xr-x 2 pi pi  4096 Jun 12 12:35 fpga
drwxr-xr-x 3 pi pi  4096 Jun 12 12:35 interface
drwxr-xr-x 3 pi pi 12288 Jun 12 12:35 target
drwxr-xr-x 2 pi pi  4096 Jun 12 12:35 tools
```

So probably there are two problems:
1. Missing Search-path
2. Missing Board-Directory

ivankravets, 2019-06-12:
Could you try to download https://bintray.com/platformio/dl-packages/download_file?file_path=tool-openocd-linux_armv7l-2.1000.190420.tar.gz into TMP folder and run with `-s /path/to/unziped/folder`?

ludetje, 2019-06-12:
That version does the job on my machine:
```
/tmp/openocd $ ./bin/openocd -f board/st_nucleo_f4.cfg -s ./scripts
Open On-Chip Debugger 0.10.0+dev-00796-ga4ac561 (2019-04-20-17:09)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 2000 kHz
adapter_nsrst_delay: 100
none separate
srst_only separate srst_nogate srst_open_drain connect_deassert_srst
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 2000 kHz
Info : STLINK V2J34M25 (API v2) VID:PID 0483:374B
Info : Target voltage: 3.266535
Info : stm32f4x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : Listening on port 3333 for gdb connections
```

Additional Info: The Version 2.1000.190610 also had wrong access rights (311)

ivankravets, 2019-06-12:
1. Please run `pio update`
2. Please provide an output from `pio run -t upload`. Do you still see this error? PlatformIO automatically appends scripts directory. See https://github.com/platformio/platform-ststm32/blob/develop/builder/main.py#L201

ludetje, 2019-06-13:
`pio update` gives me this output:
```
$ pio update
Updating tool-scons                      @ 2.20501.7      [Up-to-date]
Updating contrib-pysite                  @ 2.27.190418    [Up-to-date]
Updating tool-pioplus                    @ 2.5.0          [Up-to-date]

Platform Manager
================
Platform ST STM32
--------
Updating ststm32                         @ 5.4.1          [Up-to-date]
Updating tool-stm32duino                 @ 1.0.1          [Up-to-date]
Updating tool-openocd                    @ 2.1000.190611  [Up-to-date]
Updating framework-arduinoststm32        @ 3.10500.190327 [Up-to-date]
Updating toolchain-gccarmnoneeabi        @ 1.70201.0      [Up-to-date]

[...]
```

`pio run -t upload` fails and gives me this output:
```
$ pio run -t upload
Processing nucleo_f411re (platform: ststm32; framework: arduino; board: nucleo_f411re)
---------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/nucleo_f411re.html
PLATFORM: ST STM32 > ST Nucleo F411RE
HARDWARE: STM32F411RET6 100MHz 128KB RAM (512KB Flash)
DEBUG: CURRENT(stlink) ON-BOARD(stlink) EXTERNAL(blackmagic, jlink)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 7 compatible libraries
Scanning dependencies...
No dependencies
Checking size .pioenvs/nucleo_f411re/firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [          ]   0.7% (used 968 bytes from 131072 bytes)
PROGRAM: [          ]   2.2% (used 11764 bytes from 524288 bytes)
Configuring upload protocol...
AVAILABLE: blackmagic, jlink, mbed, stlink
CURRENT: upload_protocol = stlink
Uploading .pioenvs/nucleo_f411re/firmware.elf
Open On-Chip Debugger 0.10.0+dev-00796-ga4ac561 (2019-04-20-17:09)
Licensed under GNU GPL v2
For bug reports, read
http://openocd.org/doc/doxygen/bugs.html
/home/pi/.platformio/packages/tool-openocd/scripts/board/st_nucleo_f4.cfg:7: Error: Can't find interface/stlink.cfg
in procedure 'script'
at file "embedded:startup.tcl", line 56
at file "/home/pi/.platformio/packages/tool-openocd/scripts/board/st_nucleo_f4.cfg", line 7
*** [upload] Error 1
============================================ [ERROR] Took 21.18 seconds ============================================
```

Seems like its the same output like my first report... Am i missing something here?

ludetje, 2019-06-13:
Hmm shouldn't it be 
```
[...] 
UPLOADERFLAGS=["-s", platform.get_package_dir("tool-openocd/scripts") or ""] 
[...]
```
instead of 
```
[...]
UPLOADERFLAGS=["-s", platform.get_package_dir("tool-openocd") or ""] 
[...]
```
?

ivankravets, 2019-06-13:
Yes, this is a bug. Please update to https://github.com/platformio/platform-ststm32/releases/tag/v5.4.2

Fixed in https://github.com/platformio/platform-ststm32/commit/4a8a170d0c458abb065f3d89e55f60145b3107cc

Thanks so much for testing and report!

closed 2019-06-13 by ivankravets

ludetje, 2019-06-14:
Works on all my machines now :) Thanks!

ivankravets, 2019-06-14:
Happy coding with PlatformIO 🤗