platformio/platformio-core #2624: Home: Could not import Arduino project

Bighoneypot, 2019-06-04:
PIO Core Call Error: "The next files/directories have been created in C:\\Users\\IoT\\Documents\\PlatformIO\\Projects\\190604-123315-nodemcuv2\r\ninclude - Put project header files here\r\nlib - Put here project specific (private) libraries\r\nsrc - Put project source files here\r\nplatformio.ini - Project Configuration File\n\nError: Invalid value: Directory \"~/Documents/Arduino/libraries\" does not exist. for option `lib_extra_dirs` in section [env:nodemcuv2]"

ivankravets, 2019-06-04:
> Directory "~/Documents/Arduino/libraries

Is it true?

ivankravets, 2019-06-04:
Please open PIO IDE Terminal, type `pio upgrade --dev`, restart VSCOde.

Should work now. Thanks for the report!

closed 2019-06-04 by ivankravets

Bighoneypot, 2019-06-04:
> Please open PIO IDE Terminal, type pio upgrade --dev, restart VSCOde

Ok, now is possible to import a .ino file.

Thank you man.

Bighoneypot, 2019-06-04:
> > Directory "~/Documents/Arduino/libraries
> 
> Is it true?

Yes

Bighoneypot, 2019-06-04:
But..
when I use platformio I can't build the sketch because (.platformio applications framework-arduinoespressif8266 libraries Wire Wire.cpp: 38: 2: error: #error Wire library is not supported on this board)

instead with the arduino I can compile an the libraries Wire is compatibile.
The libraries are the same

commit 5c8c10e7 (2019-06-04): Do not check that lib_extra_dirs exist // Resolve #2624