platformio/platformio-core #2630: Home: Could not load PIO Account information

msarker23, 2019-06-04:
PIO Core Call Error: "CorePackageManager: Installing tool-pioplus @ ^2.1.4\r\nDownloading\r\nDownloading\r\nUnpacking\r\nUnpacking\n\n[Error 5] Access is denied: 'C:\\\\Users\\\\thepe_000\\\\.platformio\\\\packages\\\\tool-pioplus\\\\pioplus.exe' \r\nPlease manually remove the file `C:\\Users\\thepe_000\\.platformio\\packages\\tool-pioplus\\pioplus.exe`\r\n[Error 5] Access is denied: 'C:\\\\Users\\\\thepe_000\\\\.platformio\\\\packages\\\\tool-pioplus' \r\nPlease manually remove the file `C:\\Users\\thepe_000\\.platformio\\packages\\tool-pioplus`\r\nError: Could not install 'tool-pioplus' with version requirements '^2.1.4' for your system 'windows_x86'.\r\n\r\nPlease try this solution -> http://bit.ly/faq-package-manager"

pfeerick, 2019-06-05:
Did you try any of the suggestions listed at http://bit.ly/faq-package-manager ? `[Error 5]` is often a permissions error - either privileges, files in use, or antivirus being overzealous.

ivankravets, 2019-06-05:
Duplicate of #2473

closed 2019-06-05 by ivankravets