platformio/platformio-core #2634: Feature request:  “Save Copy Of Project As…” option

S2Doc, 2019-06-05:
It is often desirable to create a new version of a project while also having a workable copy of the original.  It would be very useful to have an option to "Save Copy of Project As..." option that would create a duplicate project folder with copies of all appropriate environment and settings files.  The user should be able to specify the name and location for this new Project.

ivankravets, 2019-06-05:
We have this: PIO Home > Open Project > "Duplicate" button

closed 2019-06-05 by ivankravets

S2Doc, 2019-06-05:
Perfect!!   Thank you very much.  That’s exactly what I needed.

Jack

MHz000, 2023-08-21:
exactly what I was looking for for a long time

nounours18200, 2025-09-14:
After having done PIO Home > Open Project > "Duplicate" button, how do I open the duplicated project ??? I don't see how to do....
To date I copy the entire code of the existing project, go to Home->New project, then paste the code in this new project.