platformio/platformio-core #2644: Home: Could not initialize project

steklobiz, 2019-06-06:
I have Platformio/VS Code on Windows 10. During creation  of STM board project I got this message. 
P.S. 'Platformio.ini' does not have any of non-ascii characters.
I have already reinstalled VS Code but it does not help.
 
PIO Core Call Error: "<Exception instance at 0x6071b08 with str error:\n Traceback (most recent call last):\n  File \"C:\\.platformio\\packages\\contrib-pysite\\twisted\\python\\reflect.py\", line 448, in safe_str\n    return str(o)\nUnicodeEncodeError: 'ascii' codec can't encode characters in position 57-65: ordinal not in range(128)\n>"

ivankravets, 2019-06-06:
Could you try the latest development version of PIO Core 4.0? Does it work for you? See http://docs.platformio.org/en/latest/ide/vscode.html#settings and `platformio-ide.useDevelopmentPIOCore`.

steklobiz, 2019-06-07:
Thanks for reply! 
I changed "Use Development PIO  Core" value and installed development version.
The process still goes wrong. Now the message is:

PIO Core Call Error: "The next files/directories have been created in 
"C:\\Users\\РђР»РµРєСЃР°РЅРґСЂ\\Documents\\PlatformIO\\Projects\\1\r\r\ninclude - Put project header files here\r\nlib - Put here project specific (private) libraries\r\nsrc - Put project source files here\r\nplatformio.ini - Project Configuration File\r\n\n\nError: Traceback (most recent call last):\r\n File \"c:\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 82, in main\r\n cli(None, None, None)\r\n File \"c:\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 764, in __call__\r\n return self.main(*args, **kwargs)\r\n File \"c:\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 717, in main\r\n rv = self.invoke(ctx)\r\n File \"c:\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\__init__.py\", line 37, in invoke\r\n return super(PlatformioCLI, self).invoke(ctx)\r\n File \"c:\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1137, in invoke\r\n return _process_result(sub_ctx.command.invoke(sub_ctx))\r\n File \"c:\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 956, in invoke\r\n return ctx.invoke(self.callback, **ctx.params)\r\n File \"c:\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 555, in invoke\r\n return callback(*args, **kwargs)\r\n File \"c:\\.platformio\\penv\\lib\\site-packages\\click\\decorators.py\", line 17, in new_func\r\n return f(get_current_context(), *args, **kwargs)\r\n File \"c:\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\init.py\", line 107, in cli\r\n pg.generate()\r\n File \"c:\\.platformio\\penv\\lib\\site-packages\\platformio\\ide\\projectgenerator.py\", line 102, in generate\r\n tpl_vars = self._load_tplvars()\r\n File \"c:\\.platformio\\penv\\lib\\site-packages\\platformio\\ide\\projectgenerator.py\", line 55, in _load_tplvars\r\n load_project_ide_data(self.project_dir, self.env_name) or {})\r\n File \"c:\\.platformio\\penv\\lib\\site-packages\\platformio\\project\\helpers.py\", line 195, in load_project_ide_data\r\n raise result.exception\r\nUnicodeEncodeError: 'ascii' codec can't encode characters in position 9-17: ordinal not in range(128)\r\n\r\n============================================================\r\n\r\nAn unexpected error occurred. Further steps:\r\n\r\n* Verify that you have the latest version of PlatformIO using\r\n `pip install -U platformio` command\r\n\r\n* Try to find answer in FAQ Troubleshooting section\r\n https://docs.platformio.org/page/faq.html\r\n\r\n* Report this problem to the developers\r\n https://github.com/platformio/platformio-core/issues\r\n\r\n============================================================"

ivankravets, 2019-06-07:
Could you open PlatformIO IDE Terminal and re-run
```
pio upgrade --dev
```

Restart VSCode.

steklobiz, 2019-06-08:
yes, i maded it. Here is a message:

PIO Core Call Error: "The next files/directories have been created in C:\\Users\\РђР»РµРєСЃР°РЅРґСЂ\\Documents\\PlatformIO\\Projects\\1\r\r\ninclude - Put project header files here\r\nlib - Put here project specific (private) libraries\r\nsrc - Put project source files here\r\nplatformio.ini - Project Configuration File\r\n\n\nError: Traceback (most recent call last):\r\n File \"c:\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 82, in main\r\n cli(None, None, None)\r\n File \"c:\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 764, in __call__\r\n return self.main(*args, **kwargs)\r\n File \"c:\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 717, in main\r\n rv = self.invoke(ctx)\r\n File \"c:\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\__init__.py\", line 37, in invoke\r\n return super(PlatformioCLI, self).invoke(ctx)\r\n File \"c:\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1137, in invoke\r\n return _process_result(sub_ctx.command.invoke(sub_ctx))\r\n File \"c:\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 956, in invoke\r\n return ctx.invoke(self.callback, **ctx.params)\r\n File \"c:\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 555, in invoke\r\n return callback(*args, **kwargs)\r\n File \"c:\\.platformio\\penv\\lib\\site-packages\\click\\decorators.py\", line 17, in new_func\r\n return f(get_current_context(), *args, **kwargs)\r\n File \"c:\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\init.py\", line 107, in cli\r\n pg.generate()\r\n File \"c:\\.platformio\\penv\\lib\\site-packages\\platformio\\ide\\projectgenerator.py\", line 102, in generate\r\n tpl_vars = self._load_tplvars()\r\n File \"c:\\.platformio\\penv\\lib\\site-packages\\platformio\\ide\\projectgenerator.py\", line 55, in _load_tplvars\r\n load_project_ide_data(self.project_dir, self.env_name) or {})\r\n File \"c:\\.platformio\\penv\\lib\\site-packages\\platformio\\project\\helpers.py\", line 195, in load_project_ide_data\r\n raise result.exception\r\nUnicodeEncodeError: 'ascii' codec can't encode characters in position 9-17: ordinal not in range(128)\r\n\r\n============================================================\r\n\r\nAn unexpected error occurred. Further steps:\r\n\r\n* Verify that you have the latest version of PlatformIO using\r\n `pip install -U platformio` command\r\n\r\n* Try to find answer in FAQ Troubleshooting section\r\n https://docs.platformio.org/page/faq.html\r\n\r\n* Report this problem to the developers\r\n https://github.com/platformio/platformio-core/issues\r\n\r\n============================================================"

ivankravets, 2019-06-10:
Sorry that disturb you. Could you open PlatformIO IDE Terminal again and type `pio upgrade --dev`. Restart VSCode. Does it work now?

steklobiz, 2019-06-10:
No problem with disturbing, thanks for you work. I made everything you asked. Here is a message
PIO Core Call Error: "The next files/directories have been created in C:\\Users\\РђР»РµРєСЃР°РЅРґСЂ\\Documents\\PlatformIO\\Projects\\temp\r\r\ninclude - Put project header files here\r\nlib - Put here project specific (private) libraries\r\nsrc - Put project source files here\r\nplatformio.ini - Project Configuration File\r\n\n\nError: Traceback (most recent call last):\r\n File \"c:\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 82, in main\r\n cli(None, None, None)\r\n File \"c:\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 764, in __call__\r\n return self.main(*args, **kwargs)\r\n File \"c:\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 717, in main\r\n rv = self.invoke(ctx)\r\n File \"c:\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\__init__.py\", line 37, in invoke\r\n return super(PlatformioCLI, self).invoke(ctx)\r\n File \"c:\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1137, in invoke\r\n return _process_result(sub_ctx.command.invoke(sub_ctx))\r\n File \"c:\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 956, in invoke\r\n return ctx.invoke(self.callback, **ctx.params)\r\n File \"c:\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 555, in invoke\r\n return callback(*args, **kwargs)\r\n File \"c:\\.platformio\\penv\\lib\\site-packages\\click\\decorators.py\", line 17, in new_func\r\n return f(get_current_context(), *args, **kwargs)\r\n File \"c:\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\init.py\", line 107, in cli\r\n pg.generate()\r\n File \"c:\\.platformio\\penv\\lib\\site-packages\\platformio\\ide\\projectgenerator.py\", line 102, in generate\r\n tpl_vars = self._load_tplvars()\r\n File \"c:\\.platformio\\penv\\lib\\site-packages\\platformio\\ide\\projectgenerator.py\", line 55, in _load_tplvars\r\n load_project_ide_data(self.project_dir, self.env_name) or {})\r\n File \"c:\\.platformio\\penv\\lib\\site-packages\\platformio\\project\\helpers.py\", line 195, in load_project_ide_data\r\n raise result.exception\r\nUnicodeEncodeError: 'ascii' codec can't encode characters in position 9-17: ordinal not in range(128)\r\n\r\n============================================================\r\n\r\nAn unexpected error occurred. Further steps:\r\n\r\n* Verify that you have the latest version of PlatformIO using\r\n `pip install -U platformio` command\r\n\r\n* Try to find answer in FAQ Troubleshooting section\r\n https://docs.platformio.org/page/faq.html\r\n\r\n* Report this problem to the developers\r\n https://github.com/platformio/platformio-core/issues\r\n\r\n============================================================"

ivankravets, 2019-06-10:
Could you contact me@ikravets.com?

steklobiz, 2019-06-10:
by the way, the core version did not change after i made upgrade. is it Ok?

ivankravets, 2019-06-10:
Yes, it's OK. I didn't increment it. However `pio upgrade --dev` uses the latest upstream version from GitHub.

ivankravets, 2019-06-10:
Please re-test with `pio upgrade --dev` and restart VSCode. Thanks!

commit 686d6156 (2019-06-07): Cast env_name to string // Resolve #2644
commit 80718ebb (2019-06-10): Fix "UnicodeEncodeError: 'ascii' codec can't encode characters" // Resolve #2644
commit 9de7297d (2019-06-10): Fix "UnicodeEncodeError: 'ascii' codec can't encode characters" // Resolve #2644
closed 2019-06-10 by ivankravets