platformio/platformio-core #2646: Platformio crashes after upgrade

ikstream, 2019-06-07:
### Configuration

**Operating system**:  5.1.7-arch1-1-ARCH

**PlatformIO Version** (`platformio --version`): PlatformIO, version 4.0.0a19

### Description of problem
After running platformio upgrade i tried to build my project which failed with:

Processing pico32 (platform: espressif32; board: pico32; framework: espidf)

-------------------------------------------------------------------------------------------------------------------------------
ValueError: invalid literal for int() with base 10: '0.0':
  File "/usr/lib/python3.7/site-packages/platformio/builder/main.py", line 89:
    env = DefaultEnvironment(**DEFAULT_ENV_OPTIONS)
  File "/home/ikstream/.platformio/packages/tool-scons/script/../engine/SCons/Defaults.py", line 88:
    _default_env = SCons.Environment.Environment(*args, **kw)
  File "/home/ikstream/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 982:
    apply_tools(self, tools, toolpath)
  File "/home/ikstream/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 107:
    env.Tool(tool)
  File "/home/ikstream/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 1788:
    tool = SCons.Tool.Tool(tool, toolpath, **kw)
  File "/home/ikstream/.platformio/packages/tool-scons/script/../engine/SCons/Tool/__init__.py", line 117:
    module = self._tool_module()
  File "/home/ikstream/.platformio/packages/tool-scons/script/../engine/SCons/Tool/__init__.py", line 232:
    module = spec.loader.load_module(spec.name)
  File "<frozen importlib._bootstrap_external>", line 407:

  File "<frozen importlib._bootstrap_external>", line 907:

  File "<frozen importlib._bootstrap_external>", line 732:

  File "<frozen importlib._bootstrap>", line 265:

  File "<frozen importlib._bootstrap>", line 696:

  File "<frozen importlib._bootstrap>", line 677:

  File "<frozen importlib._bootstrap_external>", line 728:

  File "<frozen importlib._bootstrap>", line 219:

  File "/usr/lib/python3.7/site-packages/platformio/builder/tools/pioplatform.py", line 31:
    @util.memoized()
  File "/usr/lib/python3.7/site-packages/platformio/util.py", line 58:
    self.expire = int(tdmap[expire[-1]] * int(expire[:-1]))

**platformio.ini:**

[env:pico32]
platform = espressif32
board = pico32
framework = espidf
upload_port = /dev/ttyUSB*
build_flags = -DCOMPONENT_EMBED_TXTFILES=src/ota_esp32.com.pem
board_build.partitions = partitions.csv

[platformio]
src_dir = i2s_udp/main/

### Additional info
I changed the framework to arduinol, but the error occures as well

ivankravets, 2019-06-07:
Thanks for the report! Please re-test with `$ pio upgrade --dev`.

closed 2019-06-07 by ivankravets

ikstream, 2019-06-08:
Works for me, thank you. I really enjoy using your software!

ivankravets, 2019-06-08:
Happy coding with PlatformIO 🤗

commit 5f1bd286 (2019-06-07): Fix "ValueError: invalid literal for int() with base 10: '0.0'" // Resolve #2646