platformio/platformio-core #2670: Getting this error trying to create a new project -> Home: Could not initialize project

jpecor, 2019-06-14:
PIO Core Call Error: "The next files/directories have been created in /Users/jason/Documents/PlatformIO/Projects/slomo\ninclude - Put project header files here\nlib - Put here project specific (private) libraries\nsrc - Put project source files here\nplatformio.ini - Project Configuration File\n\nError: Unknown board ID 'alorium_sno'"

ivankravets, 2019-06-15:
Do you have updated platforms in PIO HOME > Platforms > Updates?

jpecor, 2019-06-17:
Hi Ivan, 

Yes I am up-to-date.  

Both alorium_xlr8 and alorium_sno are no longer in the platforms/atmelavr/boards directory.  That looks like a recent change.  Both our Snō and XLR8 boards are listed in PIO HOME > Boards if you search for Alorium.

Could something have changed in our GitHub repo that PIO doesn't like?

Thanks!
Jason

ivankravets, 2019-06-17:
See https://github.com/platformio/platform-atmelavr/tree/master/boards

It seems that you have very old dev/platform or manually removed these manifest. I recommend to uninstall all dev/platforms which you have in a system and install the latest Atmel AVR.

jpecor, 2019-06-17:
That worked.  Thanks!

closed 2019-06-17 by jpecor