platformio/platformio-core #2672: Home: Could not load boards list

WilsonOviedo, 2019-06-14:
PIO Core Call Error: "Error: Traceback (most recent call last):\r\n  File \"c:\\users\\wilson oviedo\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 120, in main\r\n    cli(None, None, None)\r\n  File \"c:\\users\\wilson oviedo\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 700, in __call__\r\n    return self.main(*args, **kwargs)\r\n  File \"c:\\users\\wilson oviedo\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 680, in main\r\n    rv = self.invoke(ctx)\r\n  File \"c:\\users\\wilson oviedo\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1027, in invoke\r\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\r\n  File \"c:\\users\\wilson oviedo\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 873, in invoke\r\n    return ctx.invoke(self.callback, **ctx.params)\r\n  File \"c:\\users\\wilson oviedo\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 508, in invoke\r\n    return callback(*args, **kwargs)\r\n  File \"c:\\users\\wilson oviedo\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\boards.py\", line 29, in cli\r\n    return _print_boards_json(query, installed)\r\n  File \"c:\\users\\wilson oviedo\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\boar

ivankravets, 2019-06-15:
Could you provide a full output?

WilsonOviedo, 2019-06-18:
[image]
PIO Core Call Error: "Error: Traceback (most recent call last):\r\n File \"c:\\users\\wilson oviedo\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 120, in main\r\n cli(None, None, None)\r\n File \"c:\\users\\wilson oviedo\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 700, in __call__\r\n return self.main(*args, **kwargs)\r\n File \"c:\\users\\wilson oviedo\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 680, in main\r\n rv = self.invoke(ctx)\r\n File \"c:\\users\\wilson oviedo\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1027, in invoke\r\n return _process_result(sub_ctx.command.invoke(sub_ctx))\r\n File \"c:\\users\\wilson oviedo\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 873, in invoke\r\n return ctx.invoke(self.callback, **ctx.params)\r\n File \"c:\\users\\wilson oviedo\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 508, in invoke\r\n return callback(*args, **kwargs)\r\n File \"c:\\users\\wilson oviedo\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\boards.py\", line 29, in cli\r\n return _print_boards_json(query, installed)\r\n File \"c:\\users\\wilson oviedo\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\boards.py\", line 81, in _print_boards_json\r\n for board in _get_boards(installed):\r\n File \"c:\\users\\wilson oviedo\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\boards.py\", line 76, in _get_boards\r\n return pm.get_installed_boards() if installed else pm.get_all_boards()\r\n File \"c:\\users\\wilson oviedo\\.platformio\\penv\\lib\\site-packages\\platformio\\managers\\platform.py\", line 184, in get_all_boards\r\n for board in self.get_registered_boards():\r\n File \"c:\\users\\wilson oviedo\\.platformio\\penv\\lib\\site-packages\\platformio\\util.py\", line 138, in wrapper\r\n self.cache[key] = (time.time(), func(*args, **kwargs))\r\n File \"c:\\users\\wilson oviedo\\.platformio\\penv\\lib\\site-packages\\platformio\\managers\\platform.py\", line 178, in get_registered_boards\r\n return util.get_api_result(\"/boards\", cache_valid=\"7d\")\r\n File \"c:\\users\\wilson oviedo\\.platformio\\penv\\lib\\site-packages\\platformio\\util.py\", line 655, in get_api_result\r\n return json.loads(result)\r\n File \"C:\\Users\\Wilson Oviedo\\.platformio\\python27\\Lib\\json\\__init__.py\", line 339, in loads\r\n return _default_decoder.decode(s)\r\n File \"C:\\Users\\Wilson Oviedo\\.platformio\\python27\\Lib\\json\\decoder.py\", line 364, in decode\r\n obj, end = self.raw_decode(s, idx=_w(s, 0).end())\r\n File \"C:\\Users\\Wilson Oviedo\\.platformio\\python27\\Lib\\json\\decoder.py\", line 382, in raw_decode\r\n raise ValueError(\"No JSON object could be decoded\")\r\nValueError: No JSON object could be decoded\r\n\r\n============================================================\r\n\r\nAn unexpected error occurred. Further steps:\r\n\r\n* Verify that you have the latest version of PlatformIO using\r\n `pip install -U platformio` command\r\n\r\n* Try to find answer in FAQ Troubleshooting section\r\n https://docs.platformio.org/page/faq.html\r\n\r\n* Report this problem to the developers\r\n https://github.com/platformio/platformio-core/issues\r\n\r\n============================================================"

You mean this output?

ivankravets, 2019-06-21:
Could you "Update All" packages and try to init project again? See left side bar > PlatformIO > Quick Acess > Updates > Update All.

closed 2019-06-24 by ivankravets