platformio/platformio-core #2679: Home: Could not load installed platforms

LeonidKokhnovich, 2019-06-17:
PIO Core Call Error: "Error: Traceback (most recent call last):\n  File \"/Users/leonid/.platformio/penv/lib/python2.7/site-packages/platformio/__main__.py\", line 120, in main\n    cli(None, None, None)\n  File \"/Users/leonid/.platformio/penv/lib/python2.7/site-packages/click/core.py\", line 700, in __call__\n    return self.main(*args, **kwargs)\n  File \"/Users/leonid/.platformio/penv/lib/python2.7/site-packages/click/core.py\", line 680, in main\n    rv = self.invoke(ctx)\n  File \"/Users/leonid/.platformio/penv/lib/python2.7/site-packages/click/core.py\", line 1027, in invoke\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\n  File \"/Users/leonid/.platformio/penv/lib/python2.7/site-packages/click/core.py\", line 1027, in invoke\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\n  File \"/Users/leonid/.platformio/penv/lib/python2.7/site-packages/click/core.py\", line 873, in invoke\n    return ctx.invoke(self.callback, **ctx.params)\n  File \"/Users/leonid/.platformio/penv/lib/python2.7/site-packages/click/core.py\", line 508, in invoke\n    return callback(*args, **kwargs)\n  File \"/Users/leonid/.platformio/penv/lib/python2.7/site-packages/platformio/commands/platform.py\", line 223, in platform_list\n    expose_packages=False))\n  File \"/Users/leonid/.platformio/penv/lib/python2.7/site-packages/platformio/commands/platform.py\", line 79, in _get_installed_platform_data\n    p = PlatformFactory.newPlatform(platform)\n  File \"/Users/leonid/.platformio/penv/lib/python2.7/site-packages/platformio/managers/platform.py\", line 250, in newPlatform\n    _instance = platform_cls(join(platform_dir, \"platform.json\"))\n  File \"/Users/leonid/.platformio/penv/lib/python2.7/site-packages/platformio/managers/platform.py\", line 460, in __init__\n    self._manifest = util.load_json(manifest_path)\n  File \"/Users/leonid/.platformio/penv/lib/python2.7/site-packages/platformio/util.py\", line 185, in load_json\n    with open(file_path, \"r\") as f:\nIOError: [Errno 2] No such file or directory: '/Users/leonid/.platformio/platforms/arduino-esp32/platform.json'\n\n============================================================\n\nAn unexpected error occurred. Further steps:\n\n* Verify that you have the latest version of PlatformIO using\n  `pip install -U platformio` command\n\n* Try to find answer in FAQ Troubleshooting section\n  https://docs.platformio.org/page/faq.html\n\n* Report this problem to the developers\n  https://github.com/platformio/platformio-core/issues\n\n============================================================"

LeonidKokhnovich, 2019-06-17:
Removing ~/.platformio and reinstalling PlatformIO from Visual Studio Code helped to resolve the issue. Please, feel free to close the issue if it's not getting reported by other Github users. Thanks for a great development tool!

ivankravets, 2019-06-21:
> arduino-esp32/platform.json

This is not a valid dev/platform. THis is a framework.

closed 2019-06-21 by ivankravets