platformio/platformio-core #2685: Home: Could not load boards list

Carl-Philippe, 2019-06-19:
PIO Core Call Error: "Error: Traceback (most recent call last):\r\n  File \"c:\\users\\utilisateur\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 120, in main\r\n    cli(None, None, None)\r\n  File \"c:\\users\\utilisateur\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 700, in __call__\r\n    return self.main(*args, **kwargs)\r\n  File \"c:\\users\\utilisateur\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 680, in main\r\n    rv = self.invoke(ctx)\r\n  File \"c:\\users\\utilisateur\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1027, in invoke\r\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\r\n  File \"c:\\users\\utilisateur\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 873, in invoke\r\n    return ctx.invoke(self.callback, **ctx.params)\r\n  File \"c:\\users\\utilisateur\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 508, in invoke\r\n    return callback(*args, **kwargs)\r\n  File \"c:\\users\\utilisateur\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\boards.py\", line 29, in cli\r\n    return _print_boards_json(query, installed)\r\n  File \"c:\\users\\utilisateur\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\boards.py\", line 81, in _print_boards_json\r\n    for board in _get_boards(installed):\r\n  File \"c:\\users\\utilisateur\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\boards.py\", line 76, in _get_boards\r\n    return pm.get_installed_boards() if installed else pm.get_all_boards()\r\n  File \"c:\\users\\utilisateur\\.platformio\\penv\\lib\\site-packages\\platformio\\managers\\platform.py\", line 181, in get_all_boards\r\n    boards = self.get_installed_boards()\r\n  File \"c:\\users\\utilisateur\\.platformio\\penv\\lib\\site-packages\\platformio\\util.py\", line 138, in wrapper\r\n    self.cache[key] = (time.time(), func(*args, **kwargs))\r\n  File \"c:\\users\\utilisateur\\.platformio\\penv\\lib\\site-packages\\platformio\\managers\\platform.py\", line 169, in get_installed_boards\r\n    for config in p.get_boards().values():\r\n  File \"C:\\Users\\Utilisateur\\.platformio\\platforms\\atmelsam\\platform.py\", line 65, in get_boards\r\n    result[key] = self._add_default_debug_tools(result[key])\r\n  File \"C:\\Users\\Utilisateur\\.platformio\\platforms\\atmelsam\\platform.py\", line 108, in _add_default_debug_tools\r\n    assert openocd_chipname\r\nAssertionError\r\n\r\n============================================================\r\n\r\nAn unexpected error occurred. Further steps:\r\n\r\n* Verify that you have the latest version of PlatformIO using\r\n  `pip install -U platformio` command\r\n\r\n* Try to find answer in FAQ Troubleshooting section\r\n  https://docs.platformio.org/page/faq.html\r\n\r\n* Report this problem to the developers\r\n  https://github.com/platformio/platformio-core/issues\r\n\r\n============================================================"

ivankravets, 2019-06-21:
Did you create a custom board?

ivankravets, 2019-06-24:
Please reopen if you still need help

closed 2019-06-24 by ivankravets

Carl-Philippe, 2019-06-26:
I am using the Adafruit Feather M0 WiFi board, the problem diseapeared as it appeared, without any more information!
Thank you.