platformio/platformio-core #2693: PlatformIO with Visual Studio 2019 (VS2019)

MarcinPaszek, 2019-06-23:
### Configuration

**Operating system**:win 10

**PlatformIO Version** (`platformio --version`):3.6.7

### Description of problem
I have recently installed VS2019 and then I found about PlatformIO. I have read that it also supports Visual Studio IDE, not only the VSC. So I gave it a try and followed the instructions on the PlatformIO website. The problem is that platform IO does not generate any files and the project cannot be buil successfully. Reinitializing the project using platformio init does not work either. While trying to build the project, I see what is presented on the screenshot below. I have tried it for he STM32 Blue Pill and Arduino Leonardo. I am creating the project by platformIO commands by the command console.

#### Steps to Reproduce

1. Install PlatformIO
2. Install VS2019
3. Create a PlatformIO project using platformio --ini
4. Open project in VS2019
5. Reinitialize project because there are no files apart from folders
6. Try to build project

### Actual Results
[image]

### Expected Results
A build project with src files etc.

valeros, 2019-06-24:
Hi @MarcinPaszek !
If there are no files in the project directory, PlatformIO will generate an empty project and, obviously, this project wouldn't compile. If you initialized an empty project you'd need to add new files either by using VS dialog window (please make sure you specified the correct location to the new file) or by adding a new file manually and reinitializing the project. I tried both options and they work well.
Let me know if I'm missing something.

MarcinPaszek, 2019-06-24:
Does this mean that I have to add every header and library manually? Like every HAL library, math libraries etc?

valeros, 2019-06-24:
You only need to add a source file with your main program, everything else will be handled by PlatformIO (reinit might be required to update intellisense DB).

MarcinPaszek, 2019-06-24:
Thank you, I will probably try it out tomorrow and leave feedback.

MarcinPaszek, 2019-06-25:
@valeros I think I got it to work. Although I think theres another issue. As you can see, the folders in the project from my previous screen are called "Header Files" and "Source Files". In the project directory, PlaformIO creates folders with names as on screen below:
[image]
I renamed the folders in the IDE, added an example main.cpp, and the project build successfully. Without renaming the folders, there was an error because the there were no sourcefiles found in the right folder. The IDE view looks now as below:
[image]
Later I will test this project on hardware and leave further feedback. Thank you for your help.

valeros, 2019-06-25:
How did you add your `main.cpp`? The folder names in the VS project tree shouldn't be the issue, as it's only the internal VS representation. The compilation process is done in PlatformIO, so there should be source files in `src` folder one way or another.

MarcinPaszek, 2019-06-25:
I right-clicked "Source Files" and selected Add -> New Item...
This way, the source file ended up here, not in the src directory:
[image]

valeros, 2019-06-25:
As I wrote above and as it's depicted in the tutorial, please make sure that you specify the correct location to the `src` folder:
[image]

MarcinPaszek, 2019-07-02:
Sorry for my late reply. With the corrections, the project now builds fine. I just have one more question. What would be the best solution to debug this project now within Visual Studio? When I just hit Debug, the following error occurs: 
[image]
I guess it's because the IDE tries to debug this code within Windows. Do I need Visual GDB for this, or can I debug it somehow using PlatformIO?

valeros, 2019-07-02:
@MarcinPaszek Debugging your firmware in Visual Studio is not supported at the moment. Highly recommend you to try Visual Studio Code instead.

closed 2019-07-02 by MarcinPaszek