platformio/platformio-core #2694: Home: Could not install platform

asiongtobing, 2019-06-24:
PIO Core Call Error: "PlatformManager: Installing timsp430\nDownloading\nDownloading\nUnpacking\nUnpacking\ntimsp430 @ 2.0.6 has been successfully installed!\nPackageManager: Installing toolchain-timsp430 @ ~1.40603.0\n\nError: Could not find a version that satisfies the requirement '~1.40603.0' for your system 'linux_aarch64'"

asiongtobing, 2019-06-24:
PIO Core Call Error: "PlatformManager: Installing timsp430\nDownloading\nDownloading\nUnpacking\nUnpacking\ntimsp430 @ 2.0.6 has been successfully installed!\nPackageManager: Installing toolchain-timsp430 @ ~1.40603.0\n\nError: Could not find a version that satisfies the requirement '~1.40603.0' for your system 'linux_aarch64'"

ivankravets, 2019-06-24:
Please check that antivirus tools do not block PlatformIO package manager.

closed 2019-06-24 by ivankravets