platformio/platformio-core #2696: python get-platformio.py on Windows 10

psehorne, 2019-06-25:
What kind of issue is this?

- [ ] **PlatformIO Core**.

Command executed at Windows command prompt with Administrator priviledge:
      E:\>C:\Programming\python27\python get-platformio.py

RESULT:

==> Fixing Windows %PATH% Environment ...
[SUCCESS]

==> Installing Python Package Manager ...
<urlopen error [Errno 1] _ssl.c:499: error:1407742E:SSL routines:SSL23_GET_SERVER_HELLO:tlsv1 alert protocol version>
[FAILURE]
The installation process has been FAILED!
Please report about this problem here
< https://github.com/platformio/platformio-core/issues >

pfeerick, 2019-06-25:
Check what version of python 2.7 you're running... the latest (2.7.16) works for me. It appears to be a known issue for at least 2.7.6.

ivankravets, 2019-06-25:
@pfeerick you are right. This is obsolete Python or broken Python installation.

closed 2019-06-25 by ivankravets