platformio/platformio-core #2704: Home: Libraries: Builtin

TommyRakete, 2019-06-25:
PIO Core Call Error: "Error: Traceback (most recent call last):\n  File \"/Users/thomasratzlaff/.platformio/penv/lib/python2.7/site-packages/platformio/__main__.py\", line 120, in main\n    cli(None, None, None)\n  File \"/Users/thomasratzlaff/.platformio/penv/lib/python2.7/site-packages/click/core.py\", line 700, in __call__\n    return self.main(*args, **kwargs)\n  File \"/Users/thomasratzlaff/.platformio/penv/lib/python2.7/site-packages/click/core.py\", line 680, in main\n    rv = self.invoke(ctx)\n  File \"/Users/thomasratzlaff/.platformio/penv/lib/python2.7/site-packages/click/core.py\", line 1027, in invoke\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\n  File \"/Users/thomasratzlaff/.platformio/penv/lib/python2.7/site-packages/click/core.py\", line 1027, in invoke\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\n  File \"/Users/thomasratzlaff/.platformio/penv/lib/python2.7/site-packages/click/core.py\", line 873, in invoke\n    return ctx.invoke(self.callback, **ctx.params)\n  File \"/Users/thomasratzlaff/.platformio/penv/lib/python2.7/site-packages/click/core.py\", line 508, in invoke\n    return callback(*args, **kwargs)\n  File \"/Users/thomasratzlaff/.platformio/penv/lib/python2.7/site-packages/platformio/commands/lib.py\", line 287, in lib_builtin\n    items = get_builtin_libs(storage)\n  File \"/Users/thomasratzlaff/.platformio/penv/lib/python2.7/site-packages/platformio/util.py\", line 138, in wrapper\n    self.cache[key] = (time.time(), func(*args, **kwargs))\n  File \"/Users/thomasratzlaff/.platformio/penv/lib/python2.7/site-packages/platformio/managers/lib.py\", line 414, in get_builtin_libs\n    p = PlatformFactory.newPlatform(manifest['__pkg_dir'])\n  File \"/Users/thomasratzlaff/.platformio/penv/lib/python2.7/site-packages/platformio/managers/platform.py\", line 250, in newPlatform\n    _instance = platform_cls(join(platform_dir, \"platform.json\"))\n  File \"/Users/thomasratzlaff/.platformio/penv/lib/python2.7/site-packages/platformio/managers/platform.py\", line 460, in __init__\n    self._manifest = util.load_json(manifest_path)\n  File \"/Users/thomasratzlaff/.platformio/penv/lib/python2.7/site-packages/platformio/util.py\", line 185, in load_json\n    with open(file_path, \"r\") as f:\nIOError: [Errno 2] No such file or directory: '/Users/thomasratzlaff/.platformio/platforms/ITEADLIB_Arduino_Nextion/platform.json'\n\n============================================================\n\nAn unexpected error occurred. Further steps:\n\n* Verify that you have the latest version of PlatformIO using\n  `pip install -U platformio` command\n\n* Try to find answer in FAQ Troubleshooting section\n  https://docs.platformio.org/page/faq.html\n\n* Report this problem to the developers\n  https://github.com/platformio/platformio-core/issues\n\n============================================================"

TommyRakete, 2019-06-25:
...

TommyRakete, 2019-06-25:
...

ivankravets, 2019-06-26:
Please remove this folder `/Users/thomasratzlaff/.platformio/platforms/ITEADLIB_Arduino_Nextion/` and restart VSCode.

closed 2019-06-26 by ivankravets

TommyRakete, 2019-06-26:
Thanks for your help. Everything OK.