platformio/platformio-core #2709: Sparkfun Edge Cortex M4 MCU Board Support

victorromeo, 2019-06-26:
Requesting the inclusion of support for the Sparkfun Edge board with Cortex M4 MCU.

This board is quite new and the MCU is particularly low power making it ideal for ML applications.

[https://www.sparkfun.com/products/15170](url)

Is there a bounty system for new boards?  Are there open source instructions for creating new board definitions?

e3b0c442, 2019-06-27:
Would like to add some detail here.

The chip is the [Ambiq Apollo3 Blue](https://ambiqmicro.com/mcu/). Ambiq has an SDK for the chip. Sparkfun is using this chip in their new top-to-bottom Artemis platform, in addition to the Edge Cortex board above. The platform has a bare module, along with Nano, Uno, and Mega(-ish) format boards that use the same base module.

Product page: https://www.sparkfun.com/artemis

Sparkfun has created an Arduino core for this. Here is the board index file: https://raw.githubusercontent.com/sparkfun/Arduino_Boards/master/IDE_Board_Manager/package_sparkfun_index.json

Nate has offered to work with somebody to make this happen: sparkfun/Arduino_Apollo3#14 . It would be really nice if the Ambiq SDK framework was also supported, but I understand that's probably a lower priority.

kfrancis, 2019-12-17:
Any news on this?

oclyke, 2019-12-17:
We have a helpful individual who has created a PR to try to support PlatformIO in the Apollo3 Arduino core. 
[Issue](https://github.com/sparkfun/Arduino_Apollo3/issues/97)
[Pull Request](https://github.com/sparkfun/Arduino_Apollo3/pull/101)

Since I am unfamiliar with PlatformIO it would take me a while to test out the PR. If anyone who is more familiar could give it a whirl and comment on the PR that would help greatly.

jerabaul29, 2020-11-21:
Any news on this? :)

maxgerhardt, 2020-12-05:
A custom platform integration is available at https://github.com/nigelb/platform-apollo3blue

jerabaul29, 2020-12-05:
That would be amazing :) I really want to transition to these chips and having PlatformIO support them would be really great :) .

maxgerhardt, 2020-12-05:
Above integration is already a PlatformIO integration, with installation instructs in theree. Do you have the board and can test that? (:

jerabaul29, 2020-12-05:
I have the board, I try and I let you know :)

jerabaul29, 2020-12-05:
I encounter some upload_port issues, see https://github.com/nigelb/platform-apollo3blue/issues/7 , I will keep you posted.

jerabaul29, 2020-12-05:
I can fix by hand the port issues, but then I bump in some Arduino.h missing header issues. Updated the issue there. I think this is specific to this core, as I can compile with the same platformio on the same computer an Arduino Mega project that also imports Arduino.h

victorromeo, 2020-12-05:
You folk are legends. Thanks for your efforts. Keep it up.

JessicaMulein, 2020-12-08:
+1
+1 to @victorromeo

maxgerhardt, 2020-12-11:
@nigelb's platform works nicely on my Sparkfun Artemis ATP 👍 

[image]
[image]