platformio/platformio-core #2712: [CLion] CMakeListsUser.txt is not included in CMakeLists.txt

Teo-CD, 2019-06-26:
What kind of issue is this?

- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below
------------------------------------------------------------------

### Configuration

**Operating system** : Debian 9, kernel 4.9.168-1+deb9u3 x86_64

**PlatformIO Version** : 4.0.0b3

**Python Version** : 3.7.3

**CLion Version** : 2019.1.4

### Description of problem

CMakeListsUser.txt is not included in the CMakeLists.txt file thus preventing the use of custom CMake operations.

#### Steps to Reproduce

1. Init a new paltoformio project with --ide clion
2. Add a template main.cpp in src/
3. Init once more (As per the documentation).
4. Open in CLion. At this point everything should load fine and CMake will create the cmake-build-debug directory (Default in CLion)
4. Create the CMakeListsUser.txt file containing `message(${CMAKE_VERSION})` at the root of the project
5. Reload CMake project (CMake tab, reload)

### Actual Results

```
-- Configuring done
-- Generating done
```
CMakeListsUser.txt is not included.

### Expected Results

```
3.14.3
-- Configuring done
-- Generating done
```
CMakeListsUser.txt is included.

### Additional info

https://github.com/platformio/platformio-core/blob/54921c5dbdc68893fe45649d07d067818c36889b/platformio/ide/tpls/clion/CMakeLists.txt.tpl#L13-L15

From what I can make out, the `if(EXISTS CMakeListsUser.txt)` resolves relative paths from the *cmake-build-debug* to resolve relative paths, but not `include(CMakeListsUser.txt)` , which resolves from the root of the project. In the current state, touching a "CMakeListsUser.txt" file in the *cmake-build-debug* directory makes it work as expected, but this should not be needed.

#### Proposed fix

Changing `if(EXISTS CMakeListsUser.txt)` to `if(EXISTS ${CMAKE_CURRENT_SOURCE_DIR}/CMakeListsUser.txt)` looks like a clean fix to me, better than what I initially proposed in [my forum post](https://community.platformio.org/t/pio-core-4-0-0b2-clion-and-cmakeuserlists-txt/8373).

closed 2019-06-28 by ivankravets

ivankravets, 2019-06-28:
Please re-test with `pio upgrade --dev`.

Teo-CD, 2019-06-28:
Works perfectly, thanks for the quick fix ! I am really looking forward to the release of platformio 4.0 !

ivankravets, 2019-06-28:
It's planned for the beginning of next week.

commit e61caa37 (2019-06-28): Fix issue when CMakeListsUser.txt is not included // Resolve #2712