platformio/platformio-core #2716: Remove env_default warnings when default_envs is set

Niek, 2019-06-27:
PIO 4 shows a warning like this when ``env_default`` is set in the platformio.ini file, even when ``default_envs`` is set in the same file:
```
Warning! `env_default` configuration option in section [platformio] is deprecated and will be removed in the next release! Please use `default_envs` instead
```
In my opinion it would make sense not to show this warning in case ``default_envs`` is set, so it's possible to write a platformio.ini file that works in both PIO 3 and 4 without warnings.

ivankravets, 2019-06-28:
The final release of PlatformIO Core 4.0 is planned at the beginning of the next week. We will ask everyone to update to PIO Core 4.0. It has better performance and a lot of bug fixes.  It is fully compatible with Core 3.0 projects

closed 2019-06-28 by ivankravets

Niek, 2019-06-29:
I understand that PIO 4 is better and everyone should upgrade ASAP, but this is for the project maintainers who want to support both PIO 3 and 4 for the time being.

It's good that there's a warning when only ``env_default`` is set. But when both ``env_default`` and ``default_envs`` are set (in order to maintain compatibility with both PIO 3 and 4), I think this warning shouldn't be shown.

Sangra69, 2020-12-14:
I just want to get my damn 3d printer running now it won't let me. I am no expert with this pio nonsense.