platformio/platformio-core #2720: Add Shakti development board support

sathya281, 2019-06-28:
Feature Request:

We have developed a RISC V based processor called [Shakti](https://shakti.org.in/). Shakti has gained great amount of traction and the embedded/IoT community wants to try it, especially in India. So we have come up with a [software development platform](https://gitlab.com/shaktiproject/software/shakti-sdk) for our cores. We are looking for IDE's.

We find PlatformIO attractive. We would like to have Shakti devt boards supported in PlatformIO.

We have two Xilinx development boards to add namely,

1. Earty35T - shakti E class -  32 bit core
2. Carty100T - shakti C class -  64 bit core

note: We did exercises on adding sample boards and shakti-sdk to PlatformIO successfully. We also ran simple program with platformIO's debugger for our FPGA development board.

We will be happy to get onto PlatformIO at the earliest

[image]

ivankravets, 2019-06-28:
> note: We did exercises on adding sample boards and shakti-sdk to PlatformIO successfully

Could you share somewhere what you did?

sathya281, 2019-06-28:
Hi

https://gitlab.com/raghav-sec/shakti-platformio.git

This is where we host it now. Can you share the gitlab I'd of yours. I can
add you to the members list.

regards
Sathya

sathya281, 2019-06-28:
Sorry, This one

https://gitlab.com/shaktiproject/software/shakti-platformio-ide.git

ivankravets, 2019-06-28:
Page Not Found

sathya281, 2019-06-28:
Added user id "ivankravets"  to our repo. please check.
repository is too big to download

ivankravets, 2019-06-28:
Thanks! I see it now. We will create an official Shakti dev/platform soon and contact you.

sathya281, 2019-06-29:
Thanks for your reply.
We are waiting eagerly.

regards
Sathya

sathya281, 2019-07-09:
Hi,
Is there any input needed from our side ?

ivankravets, 2019-07-09:
Resolved in https://github.com/platformio/platform-shakti

Please re-test. If you have any issues, please report them to https://github.com/platformio/platform-shakti/issues

We would be thankful for any feedback.

closed 2019-07-09 by ivankravets