platformio/platformio-core #2727: Home: Libraries: Installed

viperet, 2019-06-30:
PIO Core Call Error: "No JSON object could be decoded"

ivankravets, 2019-07-03:
What is your PIO Core version?

viperet, 2019-07-03:
Im geting this error in both Atom and VSC when visiting Libreries/Installed tab.

```
$pio --version
PlatformIO, version 4.0.0rc1
```

ivankravets, 2019-07-03:
Could you re-run in terminal `pio upgrade --dev`, restart VSCode. Does it work now?

viperet, 2019-07-03:
No, getting the same error:

`PIO Core Call Error: "No JSON object could be decoded"`

ivankravets, 2019-07-03:
What do you see in a terminal when run:
```
pio lib -g list --json-output
```

viperet, 2019-07-03:
```
$ pio lib -g list --json-output
[{"__pkg_dir": "/Users/viperet/.platformio/lib/Adafruit MQTT Library_ID1092", "authors": [{"email": "info@adafruit.com", "maintainer": true, "name": "Adafruit", "url": null}], "description": "MQTT library that supports the FONA, ESP8266, Yun, and generic Arduino Client hardware.", "export": {"exclude": ["extras", "docs", "tests", "test", "*.doxyfile", "*.pdf"], "include": null}, "frameworks": ["arduino"], "homepage": null, "id": 1092, "keywords": ["communication"], "name": "Adafruit MQTT Library", "platforms": ["atmelavr", "atmelsam", "espressif32", "espressif8266", "infineonxmc", "intel_arc32", "kendryte210", "microchippic32", "nordicnrf51", "nordicnrf52", "ststm32", "ststm8", "teensy", "timsp430"], "repository": {"type": "git", "url": "https://github.com/adafruit/Adafruit_MQTT_Library"}, "version": "1.0.3"}, {"__pkg_dir": "/Users/viperet/.platformio/lib/Adafruit NeoPixel_ID28", "authors": [{"email": "info@adafruit.com", "maintainer": true, "name": "Adafruit", "url": null}], "description": "Arduino library for controlling single-wire-based LED pixels and strip.", "export": {"exclude": ["extras", "docs", "tests", "test", "*.doxyfile", "*.pdf"], "include": null}, "frameworks": ["arduino"], "homepage": null, "id": 28, "keywords": ["display"], "name": "Adafruit NeoPixel", "platforms": ["atmelavr", "atmelsam", "espressif32", "espressif8266", "infineonxmc", "intel_arc32", "kendryte210", "microchippic32", "nordicnrf51", "nordicnrf52", "ststm32", "ststm8", "teensy", "timsp430"], "repository": {"type": "git", "url": "https://github.com/adafruit/Adafruit_NeoPixel"}, "version": "1.2.3"}, {"__pkg_dir": "/Users/viperet/.platformio/lib/ArduinoJson_ID64", "authors": [{"email": null, "maintainer": false, "name": "Benoit Blanchon", "url": "https://blog.benoitblanchon.fr"}], "description": "An elegant and efficient JSON library for embedded systems", "export": {"exclude": ["fuzzing", "scripts", "test", "third-party"]}, "frameworks": ["arduino"], "homepage": "https://arduinojson.org/?utm_source=meta&utm_medium=library.json", "id": 64, "keywords": ["json", "rest", "http", "web"], "name": "ArduinoJson", "platforms": ["atmelavr", "atmelsam", "espressif32", "espressif8266", "intel_arc32", "microchippic32", "nordicnrf51", "nordicnrf52", "ststm32", "teensy", "timsp430"], "repository": {"type": "git", "url": "https://github.com/bblanchon/ArduinoJson.git"}, "version": "5.13.4"}, {"__pkg_dir": "/Users/viperet/.platformio/lib/EasyOTA_ID5731", "authors": [{"email": "regenbogencode@gmail.com", "maintainer": false, "name": "Felix Name", "url": null}], "description": "ArduinoOTA wrapper for easier use", "frameworks": ["arduino"], "id": 5731, "keywords": ["easyota", "ota"], "name": "EasyOTA", "platforms": ["espressif8266"], "repository": {"type": "git", "url": "https://gitlab.com/regenbogencode/easyota"}, "version": "59a1299a1f"}, {"__pkg_dir": "/Users/viperet/.platformio/lib/IRremoteESP8266_ID1089", "authors": [{"email": "zetoslab@gmail.com", "maintainer": false, "name": "Ken Shirriff", "url": null}, {"email": null, "maintainer": true, "name": "Mark Szabo", "url": "http://nomartini-noparty.blogspot.com/"}, {"email": null, "maintainer": true, "name": "Sebastien Warin", "url": "http://sebastien.warin.fr"}, {"email": null, "maintainer": true, "name": "David Conran", "url": "https://plus.google.com/+davidconran"}, {"email": null, "maintainer": true, "name": "Roi Dayan", "url": "https://github.com/roidayan/"}, {"email": null, "maintainer": true, "name": "Massimiliano Pinto", "url": "https://github.com/pintomax/"}], "description": "Send and receive infrared signals with multiple protocols (ESP8266)", "frameworks": ["arduino"], "id": 1089, "keywords": ["infrared", "ir", "remote", "esp8266", "esp32"], "name": "IRremoteESP8266", "platforms": ["espressif8266", "espressif32"], "repository": {"type": "git", "url": "https://github.com/markszabo/IRremoteESP8266.git"}, "version": "2.6.2"}, {"__pkg_dir": "/Users/viperet/.platformio/lib/Manchester_ID298", "authors": [{"email": null, "maintainer": false, "name": "mchr3k", "url": "https://github.com/mchr3k/arduino-libs-manchester"}], "description": "This is a Manchester encoding RF library which works on Arduino and ATTiny.", "frameworks": ["arduino"], "id": 298, "keywords": ["rf", "radio", "wireless"], "name": "Manchester", "platforms": ["atmelavr"], "repository": {"type": "git", "url": "https://github.com/mchr3k/arduino-libs-manchester.git"}, "version": "1.0"}, {"__pkg_dir": "/Users/viperet/.platformio/lib/PubSubClient_ID89", "authors": [{"email": "nick.oleary@gmail.com", "maintainer": false, "name": "Nick O'Leary", "url": "https://github.com/knolleary"}], "description": "A client library for MQTT messaging. MQTT is a lightweight messaging protocol ideal for small devices. This library allows you to send and receive MQTT messages. It supports the latest MQTT 3.1.1 protocol and can be configured to use the older MQTT 3.1 if needed. It supports all Arduino Ethernet Client compatible hardware, including the Intel Galileo/Edison, ESP8266 and TI CC3000.", "examples": "examples/*/*.ino", "export": {"exclude": "tests"}, "frameworks": ["arduino"], "id": 89, "keywords": ["ethernet", "mqtt", "m2m", "iot"], "name": "PubSubClient", "platforms": ["atmelavr", "espressif8266"], "repository": {"type": "git", "url": "https://github.com/knolleary/pubsubclient.git"}, "version": "2.7"}, {"__pkg_dir": "/Users/viperet/.platformio/lib/RadioHead_ID124", "authors": [{"email": "mikem@a [... cut ...] ": "2.1.4"}, {"__pkg_dir": "/Users/viperet/.platformio/lib/WifiManager_ID567", "authors": [{"email": null, "maintainer": false, "name": "tzapu", "url": "https://github.com/tzapu"}], "description": "ESP8266 WiFi Connection manager with fallback web configuration portal", "frameworks": ["arduino"], "id": 567, "keywords": ["wifi", "wi-fi"], "name": "WifiManager", "platforms": ["espressif8266"], "repository": {"type": "git", "url": "https://github.com/tzapu/WiFiManager.git"}, "version": "0.14"}]
```

ivankravets, 2019-07-03:
Hm... Seems like a bug. We need more details:

1. Open VSCode, close PIO Home tab
2. Open system task manager and kill all `python` processes.
3. Open PlatformIo IDE Terminal and type
```
pio home
```

Now, try to navigate to Libraries > Installed. Please paste here whole output from a console.

Thanks!

P.S: What is tour OS and Python version? (`python --version`)

viperet, 2019-07-03:
In the Chrome console:
```
16b7c663df4028997b59.min.js:49 Libraries: Installed m {message: "PIO Core Call Error", code: 4003, data: "No JSON object could be decoded", stack: "JsonRpcError: PIO Core Call Error↵    at s (http:/…0.0.1:8008/16b7c663df4028997b59.min.js:12:103541)", sagaStack: "at p ↵ JsonRpcError: PIO Core Call Error↵    at s …0.0.1:8008/16b7c663df4028997b59.min.js:12:103541)"}code: 4003data: "No JSON object could be decoded"message: "PIO Core Call Error"sagaStack: "at p ↵ JsonRpcError: PIO Core Call Error↵    at s (http://127.0.0.1:8008/16b7c663df4028997b59.min.js:12:153304)↵    at Object.o.parse (http://127.0.0.1:8008/16b7c663df4028997b59.min.js:12:155596)↵    at WebSocket.c.onmessage (http://127.0.0.1:8008/16b7c663df4028997b59.min.js:12:103541)"stack: "JsonRpcError: PIO Core Call Error↵    at s (http://127.0.0.1:8008/16b7c663df4028997b59.min.js:12:153304)↵    at Object.o.parse (http://127.0.0.1:8008/16b7c663df4028997b59.min.js:12:155596)↵    at WebSocket.c.onmessage (http://127.0.0.1:8008/16b7c663df4028997b59.min.js:12:103541)"__proto__: Errorconstructor: ƒ m(t,n,r)name: "JsonRpcError"__proto__: Object
(anonymous) @ 16b7c663df4028997b59.min.js:49
```
In the VSC console:
```
Error: {'message': 'PIO Core Call Error', 'code': 4003, 'data': 'No JSON object could be decoded'}
```

Python version: 
```
Python 2.7.10
```

I'm running MacOS Sierra 10.12.6

ivankravets, 2019-07-03:
Could you provide a TeamViewer session to debug this issue? I would be really thankful. We plan to release huge update PlatformIO Core 4.0 next week and would be good to fix all known issues.
Contact me@ikravets.com.

ivankravets, 2019-07-03:
Thanks for the remote session and report! Please open PIO IDE Terminal and type `pio upgrade --dev`. It should work now.

ivankravets, 2019-07-03:
Of course, restart VSCode after an upgrade.

commit 7c6fabae (2019-07-03): Fix an issue with unhandled warnings from PIO Core when calling it internally // Resolved #2727
closed 2019-07-03 by ivankravets