platformio/platformio-core #2733: Can't pause Debugger neither on command line nor in Eclipse

awachtler, 2019-07-02:
### Configuration

**Operating system**:

Windows10 and Linux

**PlatformIO Version** (`platformio --version`):

v4.0.0.rc1 and before

### Description of problem

If the command `pio debug ... --interface gdb` is launched, the debugger starts regularly, openocd is started via pipe and I'm able to single step and run to a breakpoint.

Nevertheless, if I let the debugger run in freerun mode (gdb-command: continue) and want to pause it after a while (e.g. to see in which endless loop my embedded programm hangs), the entire PIO is ended rather then the debugger gets halted (e.g. to evaluate a stacktrace)

PIO Version: v4.0.0rc1 and before 
Hardware: Custom SAM4E Board with Atmel-ICE / SAM70 board with FTDI2232
Debugger: openocd 

#### Steps to Reproduce

on Linux

1. run `pio debug ... --interface gdb`
2. save the current temporary gdb init file for later (~/.platformio/.chache/<tempfolder>/.pioinit => mypioinit)
3. gdb: source -v .pioinit # the debugger connects to the target and flashs the firmware
4. type "c <Enter>" until no more gdb prompt appears - the firmware is now free running
5. type "<Ctrl-C>" --> now pio exits rather then stops the debugger.

Verify that it is not the arm-none-eabi-gdb binary

1 run `~/.platformio/packages/toolchain-arm..../bin/... gdb <your-elf-file> -x mypionit `
2 repeat step 4 and 5 from above and the gdb stays in Debug Prompt.

### Additional info

I saw that in commands/debug/client.py the Debugger is launched via twisted.internet.reactor module. This module catches Ctrl-C with its own handler and terminates PIO rather then  forward it to the Debugger.

see also
https://stackoverflow.com/questions/4125772/twisted-interrupt-callback-via-keyboardinterrupt

A quick hack works at least once:
In commands/debug/client.py I replaced:
```
        print "gdb command", gdb_path, " ".join(args)
        #rv = reactor.spawnProcess(self,
        rv = self.my_spawn(self,
                        gdb_path,
                         args,
                         path=self.project_dir,
                         env=os.environ)
        return rv

    def my_spawn(self, *args, **kwargs):
        print  "****my_spawn", args, kwargs
        import subprocess
        myargs = [args[1]] + args[2][1:]
        proc = subprocess.Popen(myargs)
        proc.communicate()
```

awachtler, 2019-07-02:
Short update on my site. With a standalone gdb-wrapper script I can now start and stop debugging.

The trick is to a an empty SIGINT-handler temporarily while GDB is running ...

```python
from twisted.internet import reactor  # pylint: disable=import-error
from twisted.internet import protocol  # pylint: disable=import-error
from twisted.internet import stdio  # pylint: disable=import-error
import os
import signal

class MyPP(protocol.ProcessProtocol):

    def connectionMade(self):
        p = protocol.Protocol()
        p.dataReceived = self.onStdInData
        stdio.StandardIO(p)

    def onStdInData(self, data):
        self.transport.write(data)

    def outReceived(self, data):
        print "r>", data

    @staticmethod
    def processEnded(data):
        x = reactor.stop()
        signal.signal(signal.SIGINT, signal.default_int_handler)

def ctrl_c_received(*args, **kwargs):
    print "ctrl_c_received:", args, kwargs

if __name__ == "__main__":
    bp = MyPP()
    exe = "/home/awachtler/.platformio/packages/toolchain-gccarmnoneeabi/bin/arm-none-eabi-gdb"
    x = reactor.spawnProcess(bp , exe,
                             args = ["dummy", "myfirwmare.elf", "-x", "mypioinit"],
                             env=os.environ)
    signal.signal(signal.SIGINT, ctrl_c_received)
    reactor.run()
```

awachtler, 2019-07-02:
The signal replacement seems to work just on Linux, not (yet) on Windows ... I'll do continue on that particular issue.

awachtler, 2019-07-02:
https://bugs.python.org/issue18040 - won't fix for Py2.x :-(

ivankravets, 2019-07-04:
Thanks so much for the hints! Please re-test with the latest dev version:
```
pio upgrade --dev
```

commit caf51590 (2019-07-04): Fixed an issue when Ctrl+C(SIGINT) terminates debugging session instead of halting // Resolve #2733
closed 2019-07-04 by ivankravets