platformio/platformio-core #2734: Not able to build anything anymore

TD-er, 2019-07-02:
After upgrading PIO and/or the platformio framework packages (not sure which one caused these issues), I can no longer build anything anymore.
I already removed all framework packages in the .platformio directory, but I cannot compile anything anymore.

Any idea where to look for an option to fix this?

```
platformio run --environment normal_ESP8266_4M
Processing normal_ESP8266_4M (framework: arduino; platform: https://github.com/TD-er/platform-espressif8266.git#patch/v1.8.0_Puya; board: esp12e)
 
Error: Traceback (most recent call last):
  File "c:\users\gijs\.platformio\penv\lib\site-packages\platformio\__main__.py", line 88, in main
    cli(None, None, None)
  File "c:\users\gijs\.platformio\penv\lib\site-packages\click\core.py", line 700, in __call__

    return self.main(*args, **kwargs)
  File "c:\users\gijs\.platformio\penv\lib\site-packages\click\core.py", line 680, in main
    rv = self.invoke(ctx)
  File "c:\users\gijs\.platformio\penv\lib\site-packages\platformio\commands\__init__.py", line 37, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "c:\users\gijs\.platformio\penv\lib\site-packages\click\core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\gijs\.platformio\penv\lib\site-packages\click\core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\gijs\.platformio\penv\lib\site-packages\click\core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "c:\users\gijs\.platformio\penv\lib\site-packages\click\decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "c:\users\gijs\.platformio\penv\lib\site-packages\platformio\commands\run\command.py", line 99, in cli
    result = (envname, ep.process())
  File "c:\users\gijs\.platformio\penv\lib\site-packages\platformio\commands\run\processor.py", line 61, in process
    result = self._run_platform()
  File "c:\users\gijs\.platformio\penv\lib\site-packages\platformio\commands\run\processor.py", line 115, in _run_platform
    return p.run(build_vars, build_targets, self.silent, self.verbose)
  File "c:\users\gijs\.platformio\penv\lib\site-packages\platformio\managers\platform.py", line 373, in run
    assert isinstance(targets, list)
AssertionError
 
============================================================
 
An unexpected error occurred. Further steps:
 
* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command
 
* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html
 
* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues
 
============================================================
```

ivankravets, 2019-07-02:
Thanks, this is a bug! Let me fix it.

TD-er, 2019-07-02:
So this is supposed to work?
Should I still change the occurrences of `build_flags = ""`  into `build_flags =` ?

ivankravets, 2019-07-02:
`build_flags =` - this is correct declaration. The bug was fixed in PIO Core 4.0. Please re-test with `pio upgrade --dev`.

closed 2019-07-02 by ivankravets

ivankravets, 2019-07-02:
@TD-er it's better to leave as empty without quotes.

TD-er, 2019-07-02:
I have changed the `= ""` into '=' and also tested a build with the dev version of PIO core 4.0.0rc2
Building works fine now.

commit 69de40c4 (2019-07-02): Fix an issue when empty multiple values for project option were not casted to list // Resolve #2734