platformio/platformio-core #2748: Home: Could not load PIO Account information

n0fategear, 2019-07-05:
PIO Core Call Error

DavidPowell, 2019-07-07:
I am getting what seems to be the same error when I load PIO home from within VSCode

PIO Core Call Error: "No JSON object could be decoded".

This prevents me from logging in to this account. Possibly relevant, on this computer I have not logged in to my PIO account for many months.

ivankravets, 2019-07-09:
Thanks! Please re-test with the latest 4.0.0rc6 

```
pio upgrade --dev
```

closed 2019-07-09 by ivankravets

TD-er, 2019-07-09:
[image]

With 4.0.0r6 it is now working again. I didn't even have to perform a login anymore.

One side note. Why does PIO log me out that often?
It is rather irritating, since I then have to move over to a browser window and use LastPass to find my password.

DJManas, 2019-07-10:
Working, thanks.

ivankravets, 2019-07-10:
> Why does PIO log me out that often?

it also should be fixed. Please file a new issue if you experience this problem.

commit d003dffc (2019-07-09): Fixed an issue with broken internal call for PIO Account // Resolve #2748